RTimothyEdwards/magic #187: ext2spice doesn't respect port index when extresist is on for specific design

Sinbiosis, 2022-10-05:
### Describe the bug:

In magic, `ext2spice` doesn't respect the index assigned to a port when `ext2psice extresist on` using a specific design. This causes the .spice that is given to show the ports in the incorrect order from what is expected.

### Example:

When a given port is assigned an index, that index should also correspond to the ports placement within the spice `.subckt` header line. For example, take the following structure:

[image]

The ports currently have the following index assignment:

[image]

Performing extraction and then converting to spice with `ext2spice extresist on` gives the following incorrect header:

 ```
...
.subckt LNA VOUTP VOUTN VINN VINP INB0 INB1 VSS VDD
...
```

### To reproduce:

1. Use this attached .mag file [LNA.txt](https://github.com/RTimothyEdwards/magic/files/9718874/LNA.txt) (renamed to .txt so github would allow me to upload it, you'll need to rename it back to LNA.mag)
2. Create the incorrect spice netlist using the following commands (save the file for future comparison):
```
select top cell
ext2spice extresist on
ext2sim labels on
extract all
ext2sim
extresist
ext2spice
```
4. Now do the following to generate the correct netlist:
```
ext2spice extresist off
ext2spice
```
5. Compare the original spice netlist to the new netlist and notice how the `.subckt` header has changed.

### Notes:

I tried to use the following simpler design to test what the issue is, but the steps given above, when applied to this simpler design, work as expected. So I'm not exactly sure what the issue is.

[image]

### Environment:

Magic version 8.3 revision 328.  Compiled on Sun Oct  2 03:37:14 PM UTC 2022 using commit 189d62da9b007d4981d99bf6f391b3dcdecd63b9

Linux v5.19.0 using commit [3d7cb6b](https://github.com/torvalds/linux/commit/3d7cb6b04c3f3115719235cc6866b10326de34cd)

RTimothyEdwards, 2022-10-07:
The example has subcells, and is not reproducible without those .mag files;  could you please tarball those up and post?  I get the issue, and can probably debug it pretty quickly with a reproducible set of files.  I can probably work up a failing example myself, but I expect that you can tarball the .mag files and post them a lot faster than I can come up with another failing example.

Sinbiosis, 2022-10-08:
Yes, my apologies, I forgot the Sky130 cells are subcells in the design. Here is the tarball:
[LNA.tar.gz](https://github.com/RTimothyEdwards/magic/files/9739629/LNA.tar.gz)