RTimothyEdwards/magic #194: Floating point exception when doing gds read with GUI

growly, 2022-10-30:
I have magic 8.3.305.

I try to load a GDS file by first sourcing this .magicrc:

```
puts stdout "Sourcing design .magicrc for technology sky130A ..."

# Put grid on 0.005 pitch.  This is important, as some commands don't
# rescale the grid automatically (such as lef read?).

set scalefac [tech lambda]
if {[lindex $scalefac 1] < 2} {
    scalegrid 1 2
}

# drc off
drc euclidean on
# Change this to a fixed number for repeatable behavior with GDS writes
# e.g., "random seed 12345"
catch {random seed}

# Turn off the scale option on ext2spice or else it conflicts with the
# scale in the model files.
ext2spice scale off

# Allow override of PDK path from environment variable PDK_ROOT
# "file nativename" guards against a local PDK_ROOT with "~" in the name
if {[catch {set PDK_ROOT [file nativename $env(PDK_ROOT)]}]} {
    set PDK_ROOT /usr/local/pdk/volare/sky130/build/44a43c23c81b45b8e774ae7a84899a5a778b6b0b
}

# loading technology
tech load $PDK_ROOT/sky130A/libs.tech/magic/sky130A.tech

# load device generator
source $PDK_ROOT/sky130A/libs.tech/magic/sky130A.tcl

# load bind keys (optional)
# source $PDK_ROOT/sky130A/libs.tech/magic/sky130A-BindKeys

# set units to lambda grid 
snap lambda

# set sky130 standard power, ground, and substrate names
set VDD VPWR
set GND VGND
set SUB VSUBS

# Allow override of type of magic library views used, "mag" or "maglef",
# from environment variable MAGTYPE

if {[catch {set MAGTYPE $env(MAGTYPE)}]} {
   set MAGTYPE mag
}

# add path to reference cells
if {[file isdir ${PDK_ROOT}/sky130A/libs.ref/${MAGTYPE}]} {
    addpath ${PDK_ROOT}/sky130A/libs.ref/${MAGTYPE}/sky130_fd_pr
    addpath ${PDK_ROOT}/sky130A/libs.ref/${MAGTYPE}/sky130_fd_io
    addpath ${PDK_ROOT}/sky130A/libs.ref/${MAGTYPE}/sky130_fd_sc_hd
    addpath ${PDK_ROOT}/sky130A/libs.ref/${MAGTYPE}/sky130_fd_sc_hdll
    addpath ${PDK_ROOT}/sky130A/libs.ref/${MAGTYPE}/sky130_fd_sc_hs
    addpath ${PDK_ROOT}/sky130A/libs.ref/${MAGTYPE}/sky130_fd_sc_hvl
    addpath ${PDK_ROOT}/sky130A/libs.ref/${MAGTYPE}/sky130_fd_sc_lp
    addpath ${PDK_ROOT}/sky130A/libs.ref/${MAGTYPE}/sky130_fd_sc_ls
    addpath ${PDK_ROOT}/sky130A/libs.ref/${MAGTYPE}/sky130_fd_sc_ms
    addpath ${PDK_ROOT}/sky130A/libs.ref/${MAGTYPE}/sky130_osu_sc
    addpath ${PDK_ROOT}/sky130A/libs.ref/${MAGTYPE}/sky130_osu_sc_t18
    addpath ${PDK_ROOT}/sky130A/libs.ref/${MAGTYPE}/sky130_ml_xx_hd
    addpath ${PDK_ROOT}/sky130A/libs.ref/${MAGTYPE}/sky130_sram_macros
} else {
    addpath ${PDK_ROOT}/sky130A/libs.ref/sky130_fd_pr/${MAGTYPE}
    addpath ${PDK_ROOT}/sky130A/libs.ref/sky130_fd_io/${MAGTYPE}
    addpath ${PDK_ROOT}/sky130A/libs.ref/sky130_fd_sc_hd/${MAGTYPE}
    addpath ${PDK_ROOT}/sky130A/libs.ref/sky130_fd_sc_hdll/${MAGTYPE}
    addpath ${PDK_ROOT}/sky130A/libs.ref/sky130_fd_sc_hs/${MAGTYPE}
    addpath ${PDK_ROOT}/sky130A/libs.ref/sky130_fd_sc_hvl/${MAGTYPE}
    addpath ${PDK_ROOT}/sky130A/libs.ref/sky130_fd_sc_lp/${MAGTYPE}
    addpath ${PDK_ROOT}/sky130A/libs.ref/sky130_fd_sc_ls/${MAGTYPE}
    addpath ${PDK_ROOT}/sky130A/libs.ref/sky130_fd_sc_ms/${MAGTYPE}
    addpath ${PDK_ROOT}/sky130A/libs.ref/sky130_osu_sc/${MAGTYPE}
    addpath ${PDK_ROOT}/sky130A/libs.ref/sky130_osu_sc_t18/${MAGTYPE}
    addpath ${PDK_ROOT}/sky130A/libs.ref/sky130_ml_xx_hd/${MAGTYPE}
    addpath ${PDK_ROOT}/sky130A/libs.ref/sky130_sram_macros/${MAGTYPE}
}

# add path to GDS cells

# add path to IP from catalog.  This procedure defined in the PDK script.
catch {magic::query_mylib_ip}
# add path to local IP from user design space.  Defined in the PDK script.
catch {magic::query_my_projects}
```

and then loading the attached `library.gds`:

```
gds read /path/to/library.gds
```

I don't know for a fact that this GDS is well-formed but it does work in klayout, "layout editor" (macos) and in gdspy.

[library.gds.zip](https://github.com/RTimothyEdwards/magic/files/9895055/library.gds.zip)

I see the same thing if I open another GDS, like the `sky130A/libs.ref/sky130_fd_sc_hd/gds/sky130_fd_sc_hd.gds` file generated for Sky130.

RTimothyEdwards, 2022-10-30:
I don't have any issue reading this GDS file.  What version of magic are you using?  On what OS?

growly, 2022-10-30:
It's debian 11.4, magic 8.3.305, and this is the startup output:

```
loading history file ... 5 events added
Use openwrapper to create a new GUI-based layout window
Use closewrapper to remove a new GUI-based layout window

Magic 8.3 revision 305 - Compiled on Sat 27 Aug 2022 11:42:25 NZST.
Starting magic under Tcl interpreter
Using Tk console window
Using TrueColor, VisualID 0x21 depth 24
Processing system .magicrc file
New windows will not have a title caption.
New windows will not have scroll bars.
New windows will not have a border.
Repainting console in magic layout window colors
Using technology "minimum", version 0.0
Root cell box:
           width x height  (   llx,  lly  ), (   urx,  ury  )  area (units^2)

microns:   1.000 x 1.000   ( 0.000,  0.000), ( 1.000,  1.000)  1.000     
lambda:        1 x 1       (     0,  0    ), (     1,  1    )  1         
Main console display active (Tcl8.6.11 / Tk8.6.11)
```

I noticed that the crash only occurs when I use the GUI. Tcl/tk version is 8.6.11 as in the output above, installed through the package manager.

When I load in the startup script (I think before any GUI-based layout windows are created?) I get this output but no crash:

```
Processing system .magicrc file
Sourcing design .magicrc for technology sky130A ...
2 Magic internal units = 1 Lambda
/home/arya/src/pdk-root/sky130A/libs.tech/magic/sky130A.tech: line 19: section version:
	Error:  Magic version 8.3.306 is required by this techfile, but this version of magic is 8.3.305.
Input style sky130(vendor): scaleFactor=2, multiplier=2
/home/arya/src/pdk-root/sky130A/libs.tech/magic/sky130A.tech: line 5121: section extract:
	Malformed line for keyword defaultsidewall.  Correct usage:
	defaultsidewall types plane capacitance
/home/arya/src/pdk-root/sky130A/libs.tech/magic/sky130A.tech: line 5137: section extract:
	Malformed line for keyword defaultsidewall.  Correct usage:
	defaultsidewall types plane capacitance
/home/arya/src/pdk-root/sky130A/libs.tech/magic/sky130A.tech: line 5158: section extract:
	Malformed line for keyword defaultsidewall.  Correct usage:
	defaultsidewall types plane capacitance
/home/arya/src/pdk-root/sky130A/libs.tech/magic/sky130A.tech: line 5186: section extract:
	Malformed line for keyword defaultsidewall.  Correct usage:
	defaultsidewall types plane capacitance
/home/arya/src/pdk-root/sky130A/libs.tech/magic/sky130A.tech: line 5219: section extract:
	Malformed line for keyword defaultsidewall.  Correct usage:
	defaultsidewall types plane capacitance
/home/arya/src/pdk-root/sky130A/libs.tech/magic/sky130A.tech: line 5256: section extract:
	Malformed line for keyword defaultsidewall.  Correct usage:
	defaultsidewall types plane capacitance
/home/arya/src/pdk-root/sky130A/libs.tech/magic/sky130A.tech: line 5332: section extract:
	Malformed line for keyword defaultsidewall.  Correct usage:
	defaultsidewall types plane capacitance
/home/arya/src/pdk-root/sky130A/libs.tech/magic/sky130A.tech: line 5338: section extract:
	Malformed line for keyword defaultsidewall.  Correct usage:
	defaultsidewall types plane capacitance
/home/arya/src/pdk-root/sky130A/libs.tech/magic/sky130A.tech: line 5354: section extract:
	Malformed line for keyword defaultsidewall.  Correct usage:
	defaultsidewall types plane capacitance
/home/arya/src/pdk-root/sky130A/libs.tech/magic/sky130A.tech: line 5375: section extract:
	Malformed line for keyword defaultsidewall.  Correct usage:
	defaultsidewall types plane capacitance
/home/arya/src/pdk-root/sky130A/libs.tech/magic/sky130A.tech: line 5403: section extract:
	Malformed line for keyword defaultsidewall.  Correct usage:
	defaultsidewall types plane capacitance
/home/arya/src/pdk-root/sky130A/libs.tech/magic/sky130A.tech: line 5436: section extract:
[... 21 lines cut ...]
	Malformed line for keyword defaultsidewall.  Correct usage:
	defaultsidewall types plane capacitance
/home/arya/src/pdk-root/sky130A/libs.tech/magic/sky130A.tech: line 5690: section extract:
	Malformed line for keyword defaultsidewall.  Correct usage:
	defaultsidewall types plane capacitance
The following types are not handled by extraction and will be treated as non-electrical types:
    ubm 
The following sections of /home/arya/src/pdk-root/sky130A/libs.tech/magic/sky130A.tech contained errors:
    version
Loading sky130A Device Generator Menu ...
Point to a window first
Scaled tech values by 2 / 1 to match internal grid scaling
New windows will not have a title caption.
New windows will not have scroll bars.
New windows will not have a border.
Repainting console in magic layout window colors
Using technology "minimum", version 0.0
Root cell box:
           width x height  (   llx,  lly  ), (   urx,  ury  )  area (units^2)

microns:   2.000 x 2.000   ( 0.000,  0.000), ( 2.000,  2.000)  4.000     
lambda:     1.00 x 1.00    (  0.00,  0.00 ), (  1.00,  1.00 )  1.00      
internal:      2 x 2       (     0,  0    ), (     2,  2    )  4         
Main console display active (Tcl8.6.11 / Tk8.6.11)
```

growly, 2022-10-30:
Maybe related to one of the drawing engines?

iirc this was my config:

```
Configuration Summary (principle requirements):

X11:          yes
Python3:      yes
OpenGL:       yes
Vector fonts: yes
Cairo:        yes
Tcl/Tk:       yes

```

growly, 2022-10-31:
I upgraded to magic 8.3.333 and the problem seems to have gone away.

closed 2022-10-31 by growly