RTimothyEdwards/magic #195: Macos: typedef redefinition with different types ('struct reg' vs 'struct _XRegion *')

cheyao, 2022-11-01:
I've installed everything according to https://github.com/RTimothyEdwards/magic/blob/master/INSTALL_MacOS.md

configure mac:
```
-----------------------------------------------------------
Configuration Summary (principle requirements):

X11:          yes
Python3:      yes
OpenGL:       no

  Cairo graphics are available so OpenGL is purely optional.
Cairo:        yes
Tcl/Tk:       yes
-----------------------------------------------------------

Use 'make' to compile and 'make install' to install.

-----------------------------------------------------------
```
and then running make -j$(sysctl -n hw.ncpu) gives this error:

`` 
gcc -g   -I/usr/local/Cellar/tcl-tk/8.6.12_1/include -Wimplicit-int -fPIC  -I/usr/local/Cellar/tcl-tk/8.6.12_1/include -I. -I..  -DCAD_DIR=\"/usr/local/lib\" -DBIN_DIR=\"/usr/local/bin\" -DTCL_DIR=\"/usr/local/lib/magic/tcl\"  -DUSE_TCL_STUBS -DUSE_TK_STUBS -DPACKAGE_NAME=\"\" -DPACKAGE_TARNAME=\"\" -DPACKAGE_VERSION=\"\" -DPACKAGE_STRING=\"\" -DPACKAGE_BUGREPORT=\"\" -DPACKAGE_URL=\"\" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNISTD_H=1 -DSIZEOF_VOID_P=8 -DSIZEOF_UNSIGNED_INT=4 -DSIZEOF_UNSIGNED_LONG=8 -DSIZEOF_UNSIGNED_LONG_LONG=8 -DSTDC_HEADERS=1 -DHAVE_SETENV=1 -DHAVE_PUTENV=1 -DHAVE_SYS_MMAN_H=1 -DHAVE_DIRENT_H=1 -DHAVE_LIMITS_H=1 -DHAVE_PATHS_H=1 -DHAVE_VA_COPY=1 -DHAVE___VA_COPY=1 -DHAVE_ZLIB=1 -DFILE_LOCKS=1 -DCALMA_MODULE=1 -DCIF_MODULE=1 -DPLOT_MODULE=1 -DLEF_MODULE=1 -DROUTE_MODULE=1 -DUSE_NEW_MACROS=1 -DVECTOR_FONTS=1 -DHAVE_LIBCAIRO=1 -DMAGIC_WRAPPER=1 -D_FORTIFY_SOURCE=0 -DMAGIC_VERSION=\"`cat ../VERSION | cut -d. -f1-2`\" -DMAGIC_REVISION=\"`cat ../VERSION | cut -d. -f3`\" -DMAGIC_COMMIT=\"`git rev-parse HEAD`\" -DGCORE=\"/usr/bin/gcore\" -DSHDLIB_EXT=\".dylib\" -DNDEBUG  -c EFread.c
gcc -g   -I/usr/local/Cellar/tcl-tk/8.6.12_1/include -Wimplicit-int -fPIC  -I/usr/local/Cellar/tcl-tk/8.6.12_1/include -I. -I..  -DCAD_DIR=\"/usr/local/lib\" -DBIN_DIR=\"/usr/local/bin\" -DTCL_DIR=\"/usr/local/lib/magic/tcl\"  -DUSE_TCL_STUBS -DUSE_TK_STUBS -DPACKAGE_NAME=\"\" -DPACKAGE_TARNAME=\"\" -DPACKAGE_VERSION=\"\" -DPACKAGE_STRING=\"\" -DPACKAGE_BUGREPORT=\"\" -DPACKAGE_URL=\"\" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNISTD_H=1 -DSIZEOF_VOID_P=8 -DSIZEOF_UNSIGNED_INT=4 -DSIZEOF_UNSIGNED_LONG=8 -DSIZEOF_UNSIGNED_LONG_LONG=8 -DSTDC_HEADERS=1 -DHAVE_SETENV=1 -DHAVE_PUTENV=1 -DHAVE_SYS_MMAN_H=1 -DHAVE_DIRENT_H=1 -DHAVE_LIMITS_H=1 -DHAVE_PATHS_H=1 -DHAVE_VA_COPY=1 -DHAVE___VA_COPY=1 -DHAVE_ZLIB=1 -DFILE_LOCKS=1 -DCALMA_MODULE=1 -DCIF_MODULE=1 -DPLOT_MODULE=1 -DLEF_MODULE=1 -DROUTE_MODULE=1 -DUSE_NEW_MACROS=1 -DVECTOR_FONTS=1 -DHAVE_LIBCAIRO=1 -DMAGIC_WRAPPER=1 -D_FORTIFY_SOURCE=0 -DMAGIC_VERSION=\"`cat ../VERSION | cut -d. -f1-2`\" -DMAGIC_REVISION=\"`cat ../VERSION | cut -d. -f3`\" -DMAGIC_COMMIT=\"`git rev-parse HEAD`\" -DGCORE=\"/usr/bin/gcore\" -DSHDLIB_EXT=\".dylib\" -DNDEBUG  -c EFantenna.c
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
In file included from EFread.c:36:
In file included from ../commands/commands.h:27:
In file included from ../database/database.h:41:
In file included from ../bplane/bplane.h:214:
../bplane/bpEnum.h:509:11: warning: using the result of an assignment as a condition without parentheses [-Wparentheses]
      if(e=bpEnumNextBINS(bpe, 0)) return e;
         ~^~~~~~~~~~~~~~~~~~~~~~~
../bplane/bpEnum.h:509:11: note: place parentheses around the assignment to silence this warning
      if(e=bpEnumNextBINS(bpe, 0)) return e;
          ^
         (                       )
../bplane/bpEnum.h:509:11: note: use '==' to turn this assignment into an equality comparison
      if(e=bpEnumNextBINS(bpe, 0)) return e;
          ^
          ==
../bplane/bpEnum.h:513:11: warning: using the result of an assignment as a condition without parentheses [-Wparentheses]
      if(e=bpEnumNextBINS(bpe, 1)) return e;
         ~^~~~~~~~~~~~~~~~~~~~~~~
../bplane/bpEnum.h:513:11: note: place parentheses around the assignment to silence this warning
      if(e=bpEnumNextBINS(bpe, 1)) return e;
          ^
         (                       )
../bplane/bpEnum.h:513:11: note: use '==' to turn this assignment into an equality comparison
      if(e=bpEnumNextBINS(bpe, 1)) return e;
          ^
          ==
../bplane/bpEnum.h:517:11: warning: using the result of an assignment as a condition without parentheses [-Wparentheses]
      if(e=bpEnumNextINBOX(bpe, 0)) return e;
         ~^~~~~~~~~~~~~~~~~~~~~~~~
../bplane/bpEnum.h:517:11: note: place parentheses around the assignment to silence this warning
      if(e=bpEnumNextINBOX(bpe, 0)) return e;
          ^
         (                        )
../bplane/bpEnum.h:517:11: note: use '==' to turn this assignment into an equality comparison
      if(e=bpEnumNextINBOX(bpe, 0)) return e;
          ^
          ==
../bplane/bpEnum.h:521:11: warning: using the result of an assignment as a condition without parentheses [-Wparentheses]
      if(e=bpEnumNextINBOX(bpe, 1)) return e;
         ~^~~~~~~~~~~~~~~~~~~~~~~~
../bplane/bpEnum.h:521:11: note: place parentheses around the assignment to silence this warning
      if(e=bpEnumNextINBOX(bpe, 1)) return e;
          ^
[... 75 lines cut ...]
          ^
         (                        )
../bplane/bpEnum.h:521:11: note: use '==' to turn this assignment into an equality comparison
      if(e=bpEnumNextINBOX(bpe, 1)) return e;
          ^
          ==
../bplane/bpEnum.h:525:11: warning: using the result of an assignment as a condition without parentheses [-Wparentheses]
      if(e=bpEnumNextHASH(bpe)) return e;
         ~^~~~~~~~~~~~~~~~~~~~
../bplane/bpEnum.h:525:11: note: place parentheses around the assignment to silence this warning
      if(e=bpEnumNextHASH(bpe)) return e;
          ^
         (                    )
../bplane/bpEnum.h:525:11: note: use '==' to turn this assignment into an equality comparison
      if(e=bpEnumNextHASH(bpe)) return e;
          ^
          ==
6 warnings and 1 error generated.
In file included from EFantenna.c:36:
../extract/extractInt.h:153:3: error: typedef redefinition with different types ('struct reg' vs 'struct _XRegion *')
} Region;
  ^
/usr/local/Cellar/tcl-tk/8.6.12_1/include/X11/Xutil.h:211:26: note: previous definition is here
typedef struct _XRegion *Region;
```

cheyao, 2022-11-01:
Ok ive fixed it, for everyone who dont know, just go to  `/usr/local/Cellar/tcl-tk/tk-version/include` which tk-version is the version, and change the X11 folder's name to whatever, and make a alias of `/opt/X11/include/X11` and move it to `/usr/local/Cellar/tcl-tk/tk-version/include` then change the alias name to X11, and you have it fixed

RTimothyEdwards, 2022-11-01:
If it is a regularly-occurring problem, though, it might make sense for me to rename the conflicting `Region` structure in magic to something else like `NetRegion` or something.

daleadamjulson, 2022-11-01:
@cheyao were you able to get magic to run correctly? After applying your fix, it appears wish8.6 crashes immediately when trying to open magic. Any thoughts?

cheyao, 2022-11-01:
Still didn't go to that step now, im stuck at graywolf because the install didn't include edit_twfiles exe, but qflow and graywolf uses it, do you know where I can get it?

cheyao, 2023-02-07:
I solved this issue this time I tried to install magic.
I found out the fix to most of the compile errors on mac is to prepend the sequence `CC=gcc-12 CXX=g++-12` before the build/configure command, like for magic we should use the command `CC=gcc-12 CXX=g++-12 ./scripts/configure_mac`
(to get gcc-12 run `brew install gcc`, the 12 might change to a newer number)

closed 2023-02-07 by cheyao

xobs, 2023-02-17:
@RTimothyEdwards I actually just ran into this problem. The issue is that `include/X11/Xlib.h` has the following definition:

```cpp
#if !defined(MAC_OSX_TK)
#   include <X11/X.h>
#endif
#ifdef MAC_OSX_TK
#   include <X11/X.h>
#   define Cursor XCursor
#   define Region XRegion
#endif
```

I renamed the struct to `MagicRegion` and that fixed the issue.

Note that I'm compiling using `--without-x` because this is in a Conda environment where there is no `keysymdef.h`. However, because the inclusion of `Xlib.h` comes from Tk, this struct is still redefined even in this case.

xobs, 2023-02-17:
cc @proppy