RTimothyEdwards/magic #200: Magic flags drc errors which cannot be seen in Kalyout

lulu9312, 2022-11-14:
I ran Magic drc as well as Klayout drc on a block. Magic flags a lot of errors which cannot be seen in Klayout. Magic flags three different kind of errors. The LV to MV Diffusion spacing < 0.27um (diff/tap.3) error  is most prominent one. 

Here's a screenshot of Klayout drc:
[image]

I also checked the spacing in Klayout for diff/tap.3 and it seems correct to me.Here's the screenshot for same:
[image]

Attaching zip file which contains drc reports, gds for the block.
[drc_gds_file.zip](https://github.com/RTimothyEdwards/magic/files/10000778/drc_gds_file.zip)

d-m-bailey, 2022-11-23:
@lulu9312 Need more information to debug. Can you provide a link to your gds?

d-m-bailey, 2022-11-23:
Caused by hierarchy that magic can't process.

use `gds flatglob *$$*` before reading.

```
% load vref_gen_nmos_with_trim
DRC style is now "drc(full)"
Loading DRC CIF style.
Deep N-well spacing to N-well < 4.5um (nwell.7)
Spacing of HV nwell to HV nwell < 2.0um (nwell.8)
% drc find
There are no errors in vref_gen_nmos_with_trim.
```