RTimothyEdwards/magic #201: Magic mangles GDS, creates met2 out of met1, claims DRC error

growly, 2022-11-18:
I generate a GDS that has metal 1, a via, and metal 2:

[image]

I load magic, read the GDS, and inspect. It has created a big metal2 box in the same place?

[image]

I export the GDS from magic and can confirm the oddity in klayout:

[image]

Not sure what's going on here - any advice?

RTimothyEdwards, 2022-11-18:
My suspicion is that it is an error in the "copyup" rules for bare contacts in the tech file.  It's also possible that using the "gds flatten true" option in magic before reading the GDS might be a workaround.  Magic handles contacts differently from other tools.

growly, 2022-11-19:
Here's the same thing happening again with the `-d XR` flag (`-d OGL` does bad things on my machine, though I thought I'd compiled it with opengl):

[image]

The unadulterated GDS is attached: [library.gds.zip](https://github.com/RTimothyEdwards/magic/files/10046036/library.gds.zip)

I can recreate on the open_pdks issue tracker if you'd like?

RTimothyEdwards, 2022-11-20:
You've made your VIA1 contact cuts too large.  The VIA1 size is 0.15um x 0.15um, not 0.17um x 0.17um.
Magic is attempting to satisfy the minimum surround rule on metal1 and metal2 on a correctly-sized via, which is why the contact comes out larger when imported into magic.

growly, 2022-11-21:
Wow, incredible. Thank you!

growly, 2022-11-21:
Wait, I made them  0.15x.15 and I still see this. Attached is an updated GDS.

[image]
(please excuse the broken routing - unless that could be part of the problem?)

[image]

[library.gds.zip](https://github.com/RTimothyEdwards/magic/files/10055656/library.gds.zip)

Interestingly it doesn't happen over here?

[image]

RTimothyEdwards, 2022-11-21:
The full set of via1 rules is that via1 size is 0.15um x 0.15um, must be surrounded on two sides by 0.055um metal1 and 0.055um metal2, and two opposite sides by 0.085um metal1 and 0.085um metal2.  Your surrounds are 0.03um and 0.06um, not 0.055um and 0.085um.

growly, 2022-11-21:
Got it. Is there a way to have magic simply report DRC errors for this instead of trying to fix it like so?

closed 2022-11-21 by growly

RTimothyEdwards, 2022-11-22:
In this case, not really.  Magic isn't really trying to "fix it".  Magic defines contacts as an area to be filled (by algorithm) with contact cuts.  To work with that definition, magic needs to have a minimum contact defined as the cut area plus the minimum amount of surrounding metal on both top and bottom.

growly, 2022-11-22:
Ok, makes sense now. Thanks!