RTimothyEdwards/magic #202: magic flags DRC error when spacing should be clear

growly, 2022-11-29:
Me again,

Magic complains that the contact here and the poly are not 0.145um apart, and that the diffusion overlap of the contact (which should be 0.23um wide, and is) is less than 65 nm.

[image]

```
Diffusion contact spacing to poly < 0.145um (CO.7 - CO.6)
N-Diffusion overlap of contact < 0.065um (CO.4)
```

But when I check the GDS in klayout, the spacings are exactly 0.145 um and 0.065 um as required. The contact is 0.230 um as I also interpret to be required so I don't think it's another instance of magic creating the contact in an unexpected way (like in 
#201 )

[image]

RTimothyEdwards, 2022-11-29:
But those values aren't the requirement _in klayout_ for _mask geometry layers_;  they're the requirement _in magic_ for _magic database layers_.  Note that the first rule is written as "(CO.7 - CO.6)";  if you go to the GF documentation, the rule CO.7 is 0.15um and is "spacing from contact to poly".  The drawn contact area in magic includes the 0.005um required overlap of metal over contact (which is rule CO.6).  Technically, the second error should say "(CO.4 - CO.6)" since the rule CO.4 for N-diffusion overlap of contact is 0.07um, and again, for the rule in magic, the required 0.005um metal1 overlap has been incorporated into the drawn contact area and so has been subtracted from the rule distance.

RTimothyEdwards, 2022-11-29:
If you want to do your layout in klayout, you should probably be using the klayout DRC checker to check it, as that will be a lot easier than trying to translate between magic's derived layer set and rules and the GDS layers and rules.

growly, 2022-11-29:
oooooohhhhhhh.

I generate layout with my own tool. I’ve been using Magic’s DRC checking as a way to learn the design rules, but I did not really understand the output. I use klayout to sanity check the GDS my tool produces.

Thanks for the explanation!

closed 2022-11-29 by growly