RTimothyEdwards/magic #205: Program received signal SIGSEGV, Segmentation fault.

Baungarten-CINVESTAV, 2022-11-30:
Hello,
I am trying to harden a configurable logic block for the new MPW-8 from efabless but I get the following error. It is strange because this error does not appear with the previous version of OpenLane (c40bcaca582bb1926062656265d49a582fb9ee3ba1be). I mean this block was hardened with this earlier version.

I would like to know if the problem is a bug or how I can fix it.
[image]

First I opened an issue at OpenLane and this is their reply, [https://github.com/The-OpenROAD-Project/OpenLane/issues/1531](url)
[image]

I will appreciate the help

RTimothyEdwards, 2022-11-30:
That stack trace makes no sense at all---it's like the database was completely corrupted.
The top screenshot says "Creating issue reproducible"---Seems like this is something I need.

RTimothyEdwards, 2022-11-30:
FYI, the problem happens earlier.  DEF read-in claims to succeed without errors but something has happened when reading in net names into labels.  I'm still tracking it down.

RTimothyEdwards, 2022-12-01:
You had some _very_ long net names in this DEF file which overran the 256-character array magic was using for writing out labels to the .mag file, which is where the crash happened.  Magic version **8.3.348** corrects this issue.

Baungarten-CINVESTAV, 2022-12-01:
So, do I have to wait for that version, or is it available? Because I see that the current version was released 17 hours ago and is version 8.3.347.

Thanks for the help and quick response

RTimothyEdwards, 2022-12-01:
It is available now.  Changes that I make during the day are pushed immediately to my git server on opencircuitdesign.com, and everything gets mirrored to github once per day at 3:00am EST.

Baungarten-CINVESTAV, 2022-12-01:
I can confirm that the problem was solved with the magic update.

closed 2022-12-01 by Baungarten-CINVESTAV