RTimothyEdwards/magic #211: when install with "make", meet with error, how to solve it

youth1forever, 2023-01-12:
/home/yy/magic/tcltk/tclmagic.c:881: multiple definition of `TxGetLinePfix'; ../textio/libtextio.o:/home/yy/magic/textio/txInput.c:1108: first defined here
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:47: tclmagic.so] Error 1
make[2]: Leaving directory '/home/yy/magic/magic'
make[1]: *** [Makefile:38: tcllibrary] Error 1
make[1]: Leaving directory '/home/yy/magic'
make: *** [Makefile:26: tcl] Error 2

RTimothyEdwards, 2023-01-12:
This is caused by compiling without Tcl/Tk support, then reconfiguring with Tcl/Tk support, and running "make" again *without first doing "make clean"*.

Do a "make clean" and recompile and it should be okay.

youth1forever, 2023-01-12:
Dear,
Thanks for your answer. But when I do "make clean"->"make"->"make install", "make" is ok, but when I do "make install", Tcl may still meet with some problem. Hope for your advice.

The error picture is in the attachment. Thanks.

https://github.com/RTimothyEdwards/magic/issues/211#issuecomment-1380436957

RTimothyEdwards, 2023-01-12:
There was no attachment.  Can you please send it?

youth1forever, 2023-01-13:
after "make clean" and then "make", the result is:
--- making Tcl shared libraries
for dir in magic net2ir tcltk; do \
        (cd $dir && make tcl-main) || exit 1; done
make[2]: Entering directory '/home/yy/magic/magic'
make[2]: Nothing to be done for 'tcl-main'.
make[2]: Leaving directory '/home/yy/magic/magic'
make[2]: Entering directory '/home/yy/magic/net2ir'
echo "Nothing to do here"
Nothing to do here
make[2]: Leaving directory '/home/yy/magic/net2ir'
make[2]: Entering directory '/home/yy/magic/tcltk'
make[2]: Nothing to be done for 'tcl-main'.
make[2]: Leaving directory '/home/yy/magic/tcltk'
make[1]: Leaving directory '/home/yy/magic'

And then if I do "make install", there is still some error.

make[2]: *** [Makefile:62: /opt/magic-8.3/lib/magic/tcl/tclmagic.so] Error 1
make[2]: Leaving directory '/home/yy/magic/magic'
make[2]: Entering directory '/home/yy/magic/net2ir'
echo "Nothing to do here"
Nothing to do here
make[2]: Leaving directory '/home/yy/magic/net2ir'
make[2]: Entering directory '/home/yy/magic/tcltk'
rm -f /opt/magic-8.3/bin/magic.sh /opt/magic-8.3/bin/magic
cp magic.sh /opt/magic-8.3/bin/magic
cp: cannot create regular file '/opt/magic-8.3/bin/magic': No such file or directory
make[2]: *** [Makefile:87: /opt/magic-8.3/bin/magic.sh] Error 1
make[2]: Leaving directory '/home/yy/magic/tcltk'
make[1]: *** [Makefile:95: install-tcl-real] Error 2
make[1]: Leaving directory '/home/yy/magic'
make: *** [Makefile:92: install-tcl] Error 2