RTimothyEdwards/magic #212: Whether Magic can transfer .sp file to .gds?

youth1forever, 2023-01-13:
[image]
I meet with problem about usage. Is there any small demo that similar to the following flow.  Whether Magic can transfer .sp file to .gds?

RTimothyEdwards, 2023-01-13:
Magic can generate a basic starting layout from a SPICE netlist if and only if there is a corresponding PDK with cell generator Tcl scripts.  These exist for the open-source PDKs `sky130` and `gf180MCU`.  For those PDKs, if you start magic with the `-rcfile <startup_script>` option, then the startup script loads the device generators, and you can use the menu option `File->Import SPICE` or use the corresponding Tcl command in the Tcl interpreter.

But, assuming that you have Librecell-layout and Librecell-lib to take care of the top and bottom steps, then magic is capable of reading GDS and generating a full R-C parasitic extracted netlist.

youth1forever, 2023-01-15:
[image]
[image]

When I run command "git submodule update",  it always block at "Cloning into..." and no response just as the second image.

RTimothyEdwards, 2023-01-15:
The main thing I see in the screenshot above is "permission denied" for writing into "/home/skywater-pdk".

But you do not need to be following those instructions.  The skywater PDK will be cloned and installed automatically when you do

```
./configure --enable-sky130-pdk
make
sudo make install
```

in the open_pdks repository.

youth1forever, 2023-01-16:
When I run "make", I still can't get access to skywater in github?
[image]