RTimothyEdwards/magic #214: .configure problem

taduongadc, 2023-01-17:
Can you help in this problem:
./configure
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking target system type... x86_64-pc-linux-gnu
checking for gcc... no
checking for cc... no
checking for cl.exe... no
configure: error: in `/home/tom/magic-master/scripts':
configure: error: no acceptable C compiler found in $PATH
See `config.log' for more details
Many thanks

RTimothyEdwards, 2023-01-17:
I may not have specified some of the prerequisites that are usually available in a Linux system but may not be included in a pared-down installation.  For the problem above, you will want to install the package `build-essential`.  It's possible there are other basic packages you will need to install.

taduongadc, 2023-01-17:
Thank you very much.  I will try and let you know.
Tuan