RTimothyEdwards/magic #220: layout window not showing on macos

cheyao, 2023-02-07:
On my mac layout window the layout window is not showing.
[image]
Any ideas on why is this happening and how can I fix it?

Here is my configure summary if you need it:
```
-----------------------------------------------------------
Configuration Summary (principle requirements):

X11:          yes
Python3:      yes
OpenGL:       yes
Vector fonts: yes
Cairo:        yes
Tcl/Tk:       yes
-----------------------------------------------------------
```
Maybe this has some relationship with that if I set the DISPLAY env variable (to whatever value, for example ":0.0" or "" (blank)) and start up magic on my mac, a text window just blinks for a milisecond and dissapiers gives a seg fault (`[1]    60107 segmentation fault  magic`).

RTimothyEdwards, 2023-02-09:
The most likely reason for issues on MacOS is that you are compiling against a version of Tcl/Tk that itself is not X11 compatible. 

Did you read the instructions in the repository top level file:  https://github.com/RTimothyEdwards/magic/blob/master/INSTALL_MacOS.md ?

cheyao, 2023-02-09:
Yes I read the instructions, all dependencies are installed. The only change I made is that I used `CC=gcc-12 CXX=g++-12 ./scripts/configure_mac` as the configure command. 

If I try to use the graphics with cairo, it says that `The graphics display couldn't be correctly initialized.` instead of the seg fault when using opengl.

And If I use the command `openwrapper`, there will be this message:
```
% openwrapper
New windows will not have a title caption.
New windows will not have scroll bars.
New windows will not have a border.
bad window path name ".layout1.pane.top.magic"
```
and a blank window appears
[image]

RTimothyEdwards, 2023-02-09:
The blank window is not surprising, given the other errors.  If there is a low-level error, then Tk will still likely generate the window frame, but the underlying pixmap will not exist.

The `graphics display couldn't be correctly initialized` is the result of routine `GrTCairoInit()` returning FALSE, which it can do for a number of reasons.  It would be instructive to run under a debugger, break on this routine, single-step through it, and determine which condition returns FALSE.

Are you able to run with the default X11 interface?

RTimothyEdwards, 2023-02-09:
The `The graphics display couldn't be correctly initialized` should have been preceded by another error message explaining why the Cairo graphics couldn't be initialized.  Are you getting such a message?

RTimothyEdwards, 2023-02-09:
I think the answer to my question about the default X11 interface is that it fails, too, given that your "blank screen" image appears to claim that you started magic with `magic -noconsole tut1` (meaning default graphics), and it said `Using NULL graphics device`.  Again, were there any other error messages preceding the `Using NULL graphics device`?

cheyao, 2023-02-09:
> Are you able to run with the default X11 interface?

Nope, running with `-d X11` just gives a seg fault.

>	The The graphics display couldn't be correctly initialized should have been preceded by another error message explaining why the Cairo graphics couldn't be initialized. Are you getting such a message?
Here is everything that displays when I use cairo. 
```
$ magic -noconsole -d cairo
Use openwrapper to create a new GUI-based layout window
Use closewrapper to remove a new GUI-based layout window

Magic 8.3 revision 365 - Compiled on Thu Feb  9 20:51:17 CET 2023.
Starting magic under Tcl interpreter
Using the terminal as the console.
No suitable visual!
The graphics display couldn't be correctly initialized.
Use '-d NULL' if you don't need graphics.
```
> Were there any other error messages preceding the Using NULL graphics device?

Nope, nothing suspicious, but here is the output:
```
$ magic -noconsole -d null
Use openwrapper to create a new GUI-based layout window
Use closewrapper to remove a new GUI-based layout window

Magic 8.3 revision 365 - Compiled on Thu Feb  9 20:51:17 CET 2023.
Starting magic under Tcl interpreter
Using the terminal as the console.
Using NULL graphics device.
Processing system .magicrc file
Using technology "minimum", version 0.0
Root cell box:
           width x height  (   llx,  lly  ), (   urx,  ury  )  area (units^2)

microns:   1.000 x 1.000   ( 0.000,  0.000), ( 1.000,  1.000)  1.000
lambda:        1 x 1       (     0,  0    ), (     1,  1    )  1
% I can type normal commands here
```

Also, Is there some option to build magic with a more detailed log to see exactly whats going wrong?

RTimothyEdwards, 2023-02-09:
Actually that helped a lot.
Your first output has `No suitable visual!`, which narrows the problem down considerably.
Running `magic -d null` or `magic -dnull` will not result in any errors, but also no graphics.  This is a batch mode.

The `No suitable visual!` is printed as the result of this statement:

```
        grTCairoVisualInfo = XGetVisualInfo(grXdpy, VisualScreenMask,
                &grtemplate, &gritems);
```

. . . But I'm not sure what to make of that.  It is the first true call to an X11 routine (`DefaultScreen()` a little bit before it is usually implemented as a macro, and probably would never return an error).