RTimothyEdwards/magic #221: gds file saved with kLayout can not be opened with magic

w32agobot, 2023-02-09:
A gds-file that was generated with OpenLane can be opened with magic. If I open this file with kLayout and execute `File -> Save`, then the layout can not be opened with magic (kLayout will still open and display the file).  @hpretl tried to solve the issue by changing the linked `.lyt` file but we could not find a solution for the issue and need assistance. 

I have attached both gds files from the video for analysis [spm.zip], magic version 8.3.364. (https://github.com/RTimothyEdwards/magic/files/10699640/spm.zip). 

Related issue report in `iic-osic-tools` with reproduction steps: https://github.com/iic-jku/iic-osic-tools/issues/6

https://user-images.githubusercontent.com/92689756/217882968-76e27371-a9f0-4ad3-b9e2-5f59577405db.mp4

w32agobot, 2023-02-09:
Inb4 I might have used the wrong command in the video to open kLayout, with `ke spm.gds`(1) kLayout shows `sky130A` instead of `default` as technology, but the issue persists. 

(1) `alias ke='klayout -e -nn $PDKPATH/libs.tech/klayout/tech/$PDK.lyt -l $PDKPATH/libs.tech/klayout/tech/$PDK.lyp'`

RTimothyEdwards, 2023-02-09:
@w32agobot :  I think you're just confused about a (probably undocumented) behavior of magic.  GDS files are "libraries", and always define one or more cells (This is different from CIF, in which layout always starts with a top-level cell).  When a GDS file is read, it is not necessarily obvious what the top level cell is, and there may be multiple top level cells.  Magic takes the interpretation that if the library name matches a cell name, then it will assume that cell is the top level cell, and it will automatically load it into the layout view.  If it doesn't match, then magic will make no assumptions about what cell is the top level, and won't load anything into the layout window.

So your problem is not that magic cannot open the layout, it's that magic does not immediately load the view into the layout window.

If you choose menu option "Options->Cell Manager", you'll see "spm" at the top.  You can select it and click on "Load", and you'll see the layout in the window.  Alternately, from the command line, if you type `cellname top` it will report the top level name `spm`, and you can view it with `load spm`.

RTimothyEdwards, 2023-02-09:
Specifically, what happens is that klayout changes the library name of the GDS to "LIB", and after that magic won't automatically display the layout when it is read in.

w32agobot, 2023-02-09:
Oooh, that explains it all. Thanks!

closed 2023-02-09 by w32agobot