RTimothyEdwards/magic #225: trying to compile and install magic but got error

paramsaini, 2023-03-10:
can someone please help me out to remove this error???

sorry for the log file I can't find it. so I pasted the entire error message after make command. 
hope you can find it.

cking for cairo_user_to_device in -lcairo... yes
configure: creating ./config.status
config.status: creating defs.mak
config.status: WARNING:  'defs.mak.in' seems to ignore the --datarootdir setting

-----------------------------------------------------------
Configuration Summary (principle requirements):

X11:          yes
Python3:      yes
OpenGL:       no

  Cairo graphics are available so OpenGL is purely optional.
Cairo:        yes
Tcl/Tk:       yes
-----------------------------------------------------------

Use 'make' to compile and 'make install' to install.

-----------------------------------------------------------

(base) paramsaini@MacBook-Pro magic % make        
--- errors and warnings logged in file make.log
--- making dependencies
rm -f */Depend
for dir in bplane cmwind commands database dbwind debug drc extflat extract graphics netmenu plow resis select sim textio tiles utils windows wiring ext2sim ext2spice calma cif plot lef garouter grouter irouter mzrouter router gcr tcltk readline lisp magic net2ir tcltk; do \
		(cd $dir && /Library/Developer/CommandLineTools/usr/bin/make depend) || exit 1; done
../rules.mak:56: Depend: No such file or directory
gcc -g   -I/usr/local/Cellar/tcl-tk/8.6.13_1/include/tcl-tk -Wimplicit-int -fPIC  -I/usr/local/Cellar/tcl-tk/8.6.13_1/include/tcl-tk -I. -I..  -DCAD_DIR=\"/Users/paramsaini/open_pdks/magic\" -DBIN_DIR=\"/usr/local/bin\" -DTCL_DIR=\"/Users/paramsaini/open_pdks/magic/magic/tcl\"  -DUSE_TCL_STUBS -DUSE_TK_STUBS -DPACKAGE_NAME=\"\" -DPACKAGE_TARNAME=\"\" -DPACKAGE_VERSION=\"\" -DPACKAGE_STRING=\"\" -DPACKAGE_BUGREPORT=\"\" -DPACKAGE_URL=\"\" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNISTD_H=1 -DSIZEOF_VOID_P=8 -DSIZEOF_UNSIGNED_INT=4 -DSIZEOF_UNSIGNED_LONG=8 -DSIZEOF_UNSIGNED_LONG_LONG=8 -DSTDC_HEADERS=1 -DHAVE_SETENV=1 -DHAVE_PUTENV=1 -DHAVE_SYS_MMAN_H=1 -DHAVE_DIRENT_H=1 -DHAVE_LIMITS_H=1 -DHAVE_PATHS_H=1 -DHAVE_VA_COPY=1 -DHAVE___VA_COPY=1 -DHAVE_ZLIB=1 -DFILE_LOCKS=1 -DCALMA_MODULE=1 -DCIF_MODULE=1 -DPLOT_MODULE=1 -DLEF_MODULE=1 -DROUTE_MODULE=1 -DUSE_NEW_MACROS=1 -DVECTOR_FONTS=1 -DHAVE_LIBCAIRO=1 -DMAGIC_WRAPPER=1 -D_FORTIFY_SOURCE=0 -DMAGIC_VERSION=\"`cat ../VERSION | cut -d. -f1-2`\" -DMAGIC_REVISION=\"`cat ../VERSION | cut -d. -f3`\" -DMAGIC_COMMIT=\"`git rev-parse HEAD`\" -DGCORE=\"/usr/bin/gcore\" -DSHDLIB_EXT=\".dylib\" -DNDEBUG -MM bpDump.c bpUtils.c bpBins.c bpEnum.c bpMain.c bpStat.c | \
	sed -e "/#/D" -e "/ \//s/ \/.*\.h//" -e "/  \\\/D" \
	> Depend
make[2]: Nothing to be done for `depend'.
../rules.mak:56: Depend: No such file or directory
gcc -g   -I/usr/local/Cellar/tcl-tk/8.6.13_1/include/tcl-tk -Wimplicit-int -fPIC  -I/usr/local/Cellar/tcl-tk/8.6.13_1/include/tcl-tk -I. -I..  -DCAD_DIR=\"/Users/paramsaini/open_pdks/magic\" -DBIN_DIR=\"/usr/local/bin\" -DTCL_DIR=\"/Users/paramsaini/open_pdks/magic/magic/tcl\"  -DUSE_TCL_STUBS -DUSE_TK_STUBS -DPACKAGE_NAME=\"\" -DPACKAGE_TARNAME=\"\" -DPACKAGE_VERSION=\"\" -DPACKAGE_STRING=\"\" -DPACKAGE_BUGREPORT=\"\" -DPACKAGE_URL=\"\" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNISTD_H=1 -DSIZEOF_VOID_P=8 -DSIZEOF_UNSIGNED_INT=4 -DSIZEOF_UNSIGNED_LONG=8 -DSIZEOF_UNSIGNED_LONG_LONG=8 -DSTDC_HEADERS=1 -DHAVE_SETENV=1 -DHAVE_PUTENV=1 -DHAVE_SYS_MMAN_H=1 -DHAVE_DIRENT_H=1 -DHAVE_LIMITS_H=1 -DHAVE_PATHS_H=1 -DHAVE_VA_COPY=1 -DHAVE___VA_COPY=1 -DHAVE_ZLIB=1 -DFILE_LOCKS=1 -DCALMA_MODULE=1 -DCIF_MODULE=1 -DPLOT_MODULE=1 -DLEF_MODULE=1 -DROUTE_MODULE=1 -DUSE_NEW_MACROS=1 -DVECTOR_FONTS=1 -DHAVE_LIBCAIRO=1 -DMAGIC_WRAPPER=1 -D_FORTIFY_SOURCE=0 -DMAGIC_VERSION=\"`cat ../VERSION | cut -d. -f1-2`\" -DMAGIC_REVISION=\"`cat ../VERSION | cut -d. -f3`\" -DMAGIC_COMMIT=\"`git rev-parse HEAD`\" -DGCORE=\"/usr/bin/gcore\" -DSHDLIB_EXT=\".dylib\" -DNDEBUG -MM CMWmain.c CMWcmmnds.c CMWundo.c CMWrgbhsv.c | \
	sed -e "/#/D" -e "/ \//s/ \/.*\.h//" -e "/  \\\/D" \
	> Depend
make[2]: Nothing to be done for `depend'.
../rules.mak:56: Depend: No such file or directory
gcc -g   -I/usr/local/Cellar/tcl-tk/8.6.13_1/include/tcl-tk -Wimplicit-int -fPIC  -I/usr/local/Cellar/tcl-tk/8.6.13_1/include/tcl-tk -I. -I..  -DCAD_DIR=\"/Users/paramsaini/open_pdks/magic\" -DBIN_DIR=\"/usr/local/bin\" -DTCL_DIR=\"/Users/paramsaini/open_pdks/magic/magic/tcl\"  -DUSE_TCL_STUBS -DUSE_TK_STUBS -DPACKAGE_NAME=\"\" -DPACKAGE_TARNAME=\"\" -DPACKAGE_VERSION=\"\" -DPACKAGE_STRING=\"\" -DPACKAGE_BUGREPORT=\"\" -DPACKAGE_URL=\"\" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNISTD_H=1 -DSIZEOF_VOID_P=8 -DSIZEOF_UNSIGNED_INT=4 -DSIZEOF_UNSIGNED_LONG=8 -DSIZEOF_UNSIGNED_LONG_LONG=8 -DSTDC_HEADERS=1 -DHAVE_SETENV=1 -DHAVE_PUTENV=1 -DHAVE_SYS_MMAN_H=1 -DHAVE_DIRENT_H=1 -DHAVE_LIMITS_H=1 -DHAVE_PATHS_H=1 -DHAVE_VA_COPY=1 -DHAVE___VA_COPY=1 -DHAVE_ZLIB=1 -DFILE_LOCKS=1 -DCALMA_MODULE=1 -DCIF_MODULE=1 -DPLOT_MODULE=1 -DLEF_MODULE=1 -DROUTE_MODULE=1 -DUSE_NEW_MACROS=1 -DVECTOR_FONTS=1 -DHAVE_LIBCAIRO=1 -DMAGIC_WRAPPER=1 -D_FORTIFY_SOURCE=0 -DMAGIC_VERSION=\"`cat ../VERSION | cut -d. -f1-2`\" -DMAGIC_REVISION=\"`cat ../VERSION | cut -d. -f3`\" -DMAGIC_COMMIT=\"`git rev-parse HEAD`\" -DGCORE=\"/usr/bin/gcore\" -DSHDLIB_EXT=\".dylib\" -DNDEBUG -MM CmdSubrs.c CmdAB.c CmdCD.c CmdE.c CmdFI.c CmdLQ.c CmdRS.c CmdTZ.c CmdWizard.c CmdAuto.c | \
	sed -e "/#/D" -e "/ \//s/ \/.*\.h//" -e "/  \\\/D" \
	> Depend
make[2]: Nothing to be done for `depend'.
../rules.mak:56: Depend: No such file or directory
gcc -g   -I/usr/local/Cellar/tcl-tk/8.6.13_1/include/tcl-tk -Wimplicit-int -fPIC  -I/usr/local/Cellar/tcl-tk/8.6.13_1/include/tcl-tk -I. -I..  -DCAD_DIR=\"/Users/paramsaini/open_pdks/magic\" -DBIN_DIR=\"/usr/local/bin\" -DTCL_DIR=\"/Users/paramsaini/open_pdks/magic/magic/tcl\"  -DUSE_TCL_STUBS -DUSE_TK_STUBS -DPACKAGE_NAME=\"\" -DPACKAGE_TARNAME=\"\" -DPACKAGE_VERSION=\"\" -DPACKAGE_STRING=\"\" -DPACKAGE_BUGREPORT=\"\" -DPACKAGE_URL=\"\" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNISTD_H=1 -DSIZEOF_VOID_P=8 -DSIZEOF_UNSIGNED_INT=4 -DSIZEOF_UNSIGNED_LONG=8 -DSIZEOF_UNSIGNED_LONG_LONG=8 -DSTDC_HEADERS=1 -DHAVE_SETENV=1 -DHAVE_PUTENV=1 -DHAVE_SYS_MMAN_H=1 -DHAVE_DIRENT_H=1 -DHAVE_LIMITS_H=1 -DHAVE_PATHS_H=1 -DHAVE_VA_COPY=1 -DHAVE___VA_COPY=1 -DHAVE_ZLIB=1 -DFILE_LOCKS=1 -DCALMA_MODULE=1 -DCIF_MODULE=1 -DPLOT_MODULE=1 -DLEF_MODULE=1 -DROUTE_MODULE=1 -DUSE_NEW_MACROS=1 -DVECTOR_FONTS=1 -DHAVE_LIBCAIRO=1 -DMAGIC_WRAPPER=1 -D_FORTIFY_SOURCE=0 -DMAGIC_VERSION=\"`cat ../VERSION | cut -d. -f1-2`\" -DMAGIC_REVISION=\"`cat ../VERSION | cut -d. -f3`\" -DMAGIC_COMMIT=\"`git rev-parse HEAD`\" -DGCORE=\"/usr/bin/gcore\" -DSHDLIB_EXT=\".dylib\" -DNDEBUG -MM DBbound.c DBcell.c DBcellbox.c DBcellcopy.c DBcellname.c DBcellsrch.c DBcellsel.c DBcellsubr.c DBconnect.c DBcount.c DBexpand.c DBio.c DBlabel.c DBlabel2.c DBpaint2.c DBpaint.c DBprop.c DBtech.c DBtcontact.c DBtechname.c DBtpaint.c DBtpaint2.c DBtechtype.c DBtiles.c DBtimestmp.c DBundo.c | \
	sed -e "/#/D" -e "/ \//s/ \/.*\.h//" -e "/  \\\/D" \
	> Depend
make[2]: Nothing to be done for `depend'.
../rules.mak:56: Depend: No such file or directory
gcc -g   -I/usr/local/Cellar/tcl-tk/8.6.13_1/include/tcl-tk -Wimplicit-int -fPIC  -I/usr/local/Cellar/tcl-tk/8.6.13_1/include/tcl-tk -I. -I..  -DCAD_DIR=\"/Users/paramsaini/open_pdks/magic\" -DBIN_DIR=\"/usr/local/bin\" -DTCL_DIR=\"/Users/paramsaini/open_pdks/magic/magic/tcl\"  -DUSE_TCL_STUBS -DUSE_TK_STUBS -DPACKAGE_NAME=\"\" -DPACKAGE_TARNAME=\"\" -DPACKAGE_VERSION=\"\" -DPACKAGE_STRING=\"\" -DPACKAGE_BUGREPORT=\"\" -DPACKAGE_URL=\"\" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNISTD_H=1 -DSIZEOF_VOID_P=8 -DSIZEOF_UNSIGNED_INT=4 -DSIZEOF_UNSIGNED_LONG=8 -DSIZEOF_UNSIGNED_LONG_LONG=8 -DSTDC_HEADERS=1 -DHAVE_SETENV=1 -DHAVE_PUTENV=1 -DHAVE_SYS_MMAN_H=1 -DHAVE_DIRENT_H=1 -DHAVE_LIMITS_H=1 -DHAVE_P [... cut ...]
          ^
          ==
In file included from CmdE.c:45:
../extract/extractInt.h:152:3: error: typedef redefinition with different types ('struct reg' vs 'struct _XRegion *')
} Region;
  ^
/usr/local/Cellar/tcl-tk/8.6.13_1/include/tcl-tk/X11/Xutil.h:277:26: note: previous definition is here
typedef struct _XRegion *Region;
                         ^
5 warnings and 1 error generated.
make[2]: *** [CmdE.o] Error 1
make[1]: *** [modules] Error 1
make: *** [tcl] Error 2
(base) paramsaini@MacBook-Pro magic %

paramsaini, 2023-03-15:
@RTimothyEdwards thanks for these changes but still I got this error after compile  ./scripts/configure_mac file then make.

Last login: Wed Mar 15 21:08:13 on ttys000
(base) paramsaini@MacBook-Pro ~ % cd magic
(base) paramsaini@MacBook-Pro magic % make
--- errors and warnings logged in file make.log
--- making dependencies
rm -f */Depend
for dir in bplane cmwind commands database dbwind debug drc extflat extract graphics netmenu plow resis select sim textio tiles utils windows wiring ext2sim ext2spice calma cif plot lef garouter grouter irouter mzrouter router gcr tcltk readline lisp magic net2ir tcltk; do \
		(cd $dir && /Library/Developer/CommandLineTools/usr/bin/make depend) || exit 1; done
../rules.mak:56: Depend: No such file or directory
gcc -g   -I/usr/local/Cellar/tcl-tk/8.6.13_1/include/tcl-tk -Wimplicit-int -fPIC  -I/usr/local/Cellar/tcl-tk/8.6.13_1/include/tcl-tk -I. -I..  -DCAD_DIR=\"/usr/local/lib\" -DBIN_DIR=\"/usr/local/bin\" -DTCL_DIR=\"/usr/local/lib/magic/tcl\"  -DUSE_TCL_STUBS -DUSE_TK_STUBS -DPACKAGE_NAME=\"\" -DPACKAGE_TARNAME=\"\" -DPACKAGE_VERSION=\"\" -DPACKAGE_STRING=\"\" -DPACKAGE_BUGREPORT=\"\" -DPACKAGE_URL=\"\" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNISTD_H=1 -DSIZEOF_VOID_P=8 -DSIZEOF_UNSIGNED_INT=4 -DSIZEOF_UNSIGNED_LONG=8 -DSIZEOF_UNSIGNED_LONG_LONG=8 -DSTDC_HEADERS=1 -DHAVE_SETENV=1 -DHAVE_PUTENV=1 -DHAVE_SYS_MMAN_H=1 -DHAVE_DIRENT_H=1 -DHAVE_LIMITS_H=1 -DHAVE_PATHS_H=1 -DHAVE_VA_COPY=1 -DHAVE___VA_COPY=1 -DHAVE_ZLIB=1 -DFILE_LOCKS=1 -DCALMA_MODULE=1 -DCIF_MODULE=1 -DPLOT_MODULE=1 -DLEF_MODULE=1 -DROUTE_MODULE=1 -DUSE_NEW_MACROS=1 -DVECTOR_FONTS=1 -DHAVE_LIBCAIRO=1 -DMAGIC_WRAPPER=1 -D_FORTIFY_SOURCE=0 -DMAGIC_VERSION=\"`cat ../VERSION | cut -d. -f1-2`\" -DMAGIC_REVISION=\"`cat ../VERSION | cut -d. -f3`\" -DMAGIC_COMMIT=\"`git rev-parse HEAD`\" -DGCORE=\"/usr/bin/gcore\" -DSHDLIB_EXT=\".dylib\" -DNDEBUG -MM bpDump.c bpUtils.c bpBins.c bpEnum.c bpMain.c bpStat.c | \
	sed -e "/#/D" -e "/ \//s/ \/.*\.h//" -e "/  \\\/D" \
	> Depend
make[2]: Nothing to be done for `depend'.
../rules.mak:56: Depend: No such file or directory
gcc -g   -I/usr/local/Cellar/tcl-tk/8.6.13_1/include/tcl-tk -Wimplicit-int -fPIC  -I/usr/local/Cellar/tcl-tk/8.6.13_1/include/tcl-tk -I. -I..  -DCAD_DIR=\"/usr/local/lib\" -DBIN_DIR=\"/usr/local/bin\" -DTCL_DIR=\"/usr/local/lib/magic/tcl\"  -DUSE_TCL_STUBS -DUSE_TK_STUBS -DPACKAGE_NAME=\"\" -DPACKAGE_TARNAME=\"\" -DPACKAGE_VERSION=\"\" -DPACKAGE_STRING=\"\" -DPACKAGE_BUGREPORT=\"\" -DPACKAGE_URL=\"\" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNISTD_H=1 -DSIZEOF_VOID_P=8 -DSIZEOF_UNSIGNED_INT=4 -DSIZEOF_UNSIGNED_LONG=8 -DSIZEOF_UNSIGNED_LONG_LONG=8 -DSTDC_HEADERS=1 -DHAVE_SETENV=1 -DHAVE_PUTENV=1 -DHAVE_SYS_MMAN_H=1 -DHAVE_DIRENT_H=1 -DHAVE_LIMITS_H=1 -DHAVE_PATHS_H=1 -DHAVE_VA_COPY=1 -DHAVE___VA_COPY=1 -DHAVE_ZLIB=1 -DFILE_LOCKS=1 -DCALMA_MODULE=1 -DCIF_MODULE=1 -DPLOT_MODULE=1 -DLEF_MODULE=1 -DROUTE_MODULE=1 -DUSE_NEW_MACROS=1 -DVECTOR_FONTS=1 -DHAVE_LIBCAIRO=1 -DMAGIC_WRAPPER=1 -D_FORTIFY_SOURCE=0 -DMAGIC_VERSION=\"`cat ../VERSION | cut -d. -f1-2`\" -DMAGIC_REVISION=\"`cat ../VERSION | cut -d. -f3`\" -DMAGIC_COMMIT=\"`git rev-parse HEAD`\" -DGCORE=\"/usr/bin/gcore\" -DSHDLIB_EXT=\".dylib\" -DNDEBUG -MM CMWmain.c CMWcmmnds.c CMWundo.c CMWrgbhsv.c | \
	sed -e "/#/D" -e "/ \//s/ \/.*\.h//" -e "/  \\\/D" \
	> Depend
make[2]: Nothing to be done for `depend'.
../rules.mak:56: Depend: No such file or directory
gcc -g   -I/usr/local/Cellar/tcl-tk/8.6.13_1/include/tcl-tk -Wimplicit-int -fPIC  -I/usr/local/Cellar/tcl-tk/8.6.13_1/include/tcl-tk -I. -I..  -DCAD_DIR=\"/usr/local/lib\" -DBIN_DIR=\"/usr/local/bin\" -DTCL_DIR=\"/usr/local/lib/magic/tcl\"  -DUSE_TCL_STUBS -DUSE_TK_STUBS -DPACKAGE_NAME=\"\" -DPACKAGE_TARNAME=\"\" -DPACKAGE_VERSION=\"\" -DPACKAGE_STRING=\"\" -DPACKAGE_BUGREPORT=\"\" -DPACKAGE_URL=\"\" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNISTD_H=1 -DSIZEOF_VOID_P=8 -DSIZEOF_UNSIGNED_INT=4 -DSIZEOF_UNSIGNED_LONG=8 -DSIZEOF_UNSIGNED_LONG_LONG=8 -DSTDC_HEADERS=1 -DHAVE_SETENV=1 -DHAVE_PUTENV=1 -DHAVE_SYS_MMAN_H=1 -DHAVE_DIRENT_H=1 -DHAVE_LIMITS_H=1 -DHAVE_PATHS_H=1 -DHAVE_VA_COPY=1 -DHAVE___VA_COPY=1 -DHAVE_ZLIB=1 -DFILE_LOCKS=1 -DCALMA_MODULE=1 -DCIF_MODULE=1 -DPLOT_MODULE=1 -DLEF_MODULE=1 -DROUTE_MODULE=1 -DUSE_NEW_MACROS=1 -DVECTOR_FONTS=1 -DHAVE_LIBCAIRO=1 -DMAGIC_WRAPPER=1 -D_FORTIFY_SOURCE=0 -DMAGIC_VERSION=\"`cat ../VERSION | cut -d. -f1-2`\" -DMAGIC_REVISION=\"`cat ../VERSION | cut -d. -f3`\" -DMAGIC_COMMIT=\"`git rev-parse HEAD`\" -DGCORE=\"/usr/bin/gcore\" -DSHDLIB_EXT=\".dylib\" -DNDEBUG -MM CmdSubrs.c CmdAB.c CmdCD.c CmdE.c CmdFI.c CmdLQ.c CmdRS.c CmdTZ.c CmdWizard.c CmdAuto.c | \
	sed -e "/#/D" -e "/ \//s/ \/.*\.h//" -e "/  \\\/D" \
	> Depend
make[2]: Nothing to be done for `depend'.
../rules.mak:56: Depend: No such file or directory
gcc -g   -I/usr/local/Cellar/tcl-tk/8.6.13_1/include/tcl-tk -Wimplicit-int -fPIC  -I/usr/local/Cellar/tcl-tk/8.6.13_1/include/tcl-tk -I. -I..  -DCAD_DIR=\"/usr/local/lib\" -DBIN_DIR=\"/usr/local/bin\" -DTCL_DIR=\"/usr/local/lib/magic/tcl\"  -DUSE_TCL_STUBS -DUSE_TK_STUBS -DPACKAGE_NAME=\"\" -DPACKAGE_TARNA [... cut ...]
            status= XAllocColorCells(grXdpy, grXcmap, TRUE, grDisplay.planes,
                    ^
grTk1.c:600:13: error: implicit declaration of function 'XCreateColormap' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
                grXcmap = XCreateColormap(grXdpy, xwind, grVisual, AllocAll);
                          ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
9 warnings and 20 errors generated.
make[2]: *** [grTk1.o] Error 1
make[1]: *** [modules] Error 1
make: *** [tcl] Error 2
(base) paramsaini@MacBook-Pro magic %

RTimothyEdwards, 2023-03-16:
It appears to be an incorrect OS setup for X11.  At the top of file `grTk1.c` is the line `#include <X11/Xlib.h>`.  On my (Fedora Linux) system, that file is at `/usr/include/X11/Xlib.h` and contains all of the function prototypes that your compiler is complaining about being missing.  I find it very odd that it does not complain about the file `Xlib.h` itself---apparently it found it---but it appears that the file does not have the usual contents, which is mostly function prototype declarations for all the Xlib functions.

RTimothyEdwards, 2023-03-16:
One clue may be the mention of `/usr/local/Cellar/tcl-tk/8.6.13_1/include/tcl-tk/X11/Xutil.h` in your output.  It seems like this version of Tcl/Tk has X11 header files.  If the same directory has an `Xlib.h` then it might be shadowing the normal `Xlib.h`.