RTimothyEdwards/magic #228: How to blackbox a specific cell in a gds for drc?

kareefardi, 2023-03-22:
I couldn't figure out how to replace a cell in a gds by its lef/maglef view - even then I am not sure if that's sufficient . This happens with sram cells in sky130. I was intending to open this a long time ago but I keep forgetting so I am opening this now in the hopes that there is a known method. If needed, I can provide an example.

RTimothyEdwards, 2023-03-22:
@kareefardi :  There are multiple ways to do this.
(1) Pre-load the maglef view of the SRAM, then read GDS with the `gds noduplicates true` option set.
(2) Read GDS normally, then do `cellname filepath <sram_cellname> <path_to_maglef_view>` and `flush <sram_cellname>`

d-m-bailey, 2023-03-23:
@RTimothyEdwards Thanks for the clarification. Accordingly, my suggestion works for LVS but not for DRC. With `property LEFview true`, it appears that connectivity extraction in preformed on all layers (but maybe not hierarchically), but devices and subcells are not recognized.