RTimothyEdwards/magic #231: load -dereference error

kareefardi, 2023-04-09:
Reading a mag file outside of the working directory with `-dereference` flag results in the following error:
```
Cell user_proj_example couldn't be read
No such file or directory
Creating new cell
```

kareefardi, 2023-04-09:
Also it appears that `-fail` has no effect with `-dereference`; magic still outputs `Creating new cell`

RTimothyEdwards, 2023-04-09:
(1) The first post sounds like behavior as intended.  `-dereference` means that magic will not enforce that a cell must be read from the directory specified for it in the parent cell's .mag file.  In that case, the behavior is to check through all the search paths.  If you want magic to search a specific directory, then you need to tell it by using the `addpath` command.

(2) The second post sounds like another known issue, which is that file loading can be deferred for subcells, and not all command options passed to "load" are still remembered by the time magic gets around to the deferred load.  The "fail" and "silent" options both need to be handled the same way as "dereference", which is to set a flag in the (unloaded) child cell, so that when the deferred load happens, the option will be remembered for that cell.

d-m-bailey, 2023-04-09:
@kareefardi Is this part of the `make ship` recipe for caravel/caravan? If it is, you might be loading the wrong version of `caravel_core`. I believe the correct version of `caravel_core` has an empty `user_project_wrapper` with no `user_proj_example`. How about after loading `caravel_core` from `<user project>/mag`, setting the top cell to `user_project_wrapper` and replacing the gds link.

kareefardi, 2023-04-10:
@d-m-bailey no this isn't about `make ship`

> (1) The first post sounds like behavior as intended. `-dereference` means that magic will not enforce that a cell must be read from the directory specified for it in the parent cell's .mag file. In [...]

(1) makes sense. (2) remains an issue, but it isn't exactly the main topic of this issue so it would be fine if this issue needs to be closed.