RTimothyEdwards/magic #232: CRITICAL: Conflicting case causes extraction of duplicate ports

d-m-bailey, 2023-04-11:
MPW-8 slot-015 uses `sram_2kbyte_1rw1r_32x512_8`.

The `openram_dff` cell has both `VDD` and `vdd` labels and `GND` and `gnd` labels on the layout.
[image]

This extracts to
```
.subckt EL_G3_sky130_fd_bd_sram__openram_dff clk Q D vdd gnd VDD GND
X0 VDD a_28_102# a_389_712# VDD sky130_fd_pr__pfet_01v8 ad=0.42 pd=3.28 as=0.315 ps=3.21 w=3 l=0.15
X1 a_47_611# clk a_197_712# VDD sky130_fd_pr__pfet_01v8 ad=0.675 pd=3.45 as=0.315 ps=3.21 w=3 l=0.15
X2 a_239_76# clk VDD VDD sky130_fd_pr__pfet_01v8 ad=0.795 pd=6.53 as=0.42 ps=3.28 w=3 l=0.15
X3 a_197_712# D VDD VDD sky130_fd_pr__pfet_01v8 ad=0.315 pd=3.21 as=0.42 ps=3.28 w=3 l=0.15
X4 QN clk a_547_102# GND sky130_fd_pr__nfet_01v8 ad=0.225 pd=1.45 as=0.105 ps=1.21 w=1 l=0.15
X5 GND Q a_739_102# GND sky130_fd_pr__nfet_01v8 ad=0.14 pd=1.28 as=0.105 ps=1.21 w=1 l=0.15
...
```
with floating `vdd` and `gnd` ports.

During LVS, the parent circuit `dff_buf_0` matches topologically but has a port mismatch.
```
Netlists match uniquely with port errors.                                         

Subcircuit pins:                                                                  
Circuit 1: EL_G3_sky130_sram_2kbyte_1rw1r_32x512_8_dff_buf_0                      |Circuit 2: dff_buf_0
----------------------------------------------------------------------------------|----------------------------------------------------------------------------------
D                                                                                 |D
clk                                                                               |clk
Q                                                                                 |Q
vdd                                                                               |vdd
EL_G3_sky130_fd_bd_sram__openram_dff_0/vdd                                        |(no matching pin)
gnd                                                                               |gnd
Qb                                                                                |Qb
---------------------------------------------------------------------------------------------------------------------------------------------------------------------
```
netgen attempts to alter the pin list to match, and gives this message.
```
Port number 7 greater than number of ports 6
Port number 7 greater than number of ports 6
Port number 7 greater than number of ports 6
Cell pin lists for EL_G3_sky130_sram_2kbyte_1rw1r_32x512_8_dff_buf_0 and dff_buf_0 altered to match.
```
At the higher levels, the `gnd` pin of the a sister cell is seen as connected to `vdd` even though it is not. `vdd` is seen as disconnected even though it is not.
```
Cell dff_buf_array_0 (1) disconnected node: dff_buf_0:dff_r0_c0/sky130_fd_bd_sram__openram_dff:dff_buf_dff/VDD
Cell dff_buf_array_0 (1) disconnected node: dff_buf_0:dff_r0_c0/pinv_2:dff_buf_inv1/vdd
Cell dff_buf_array_0 (1) disconnected node: dff_buf_0:dff_r0_c0/pinv_3:dff_buf_inv2/vdd
```
```---------------------------------------------------------------------------------------------------------------------------------------------------------------------
Net: vdd                                                                          |Net: vdd
  EL_G3_sky130_sram_2kbyte_1rw1r_32x512_8_pinv_2/vdd = 1                          |  sky130_fd_bd_sram__openram_dff/VDD = 1
  EL_G3_sky130_sram_2kbyte_1rw1r_32x512_8_pinv_3/vdd = 1                          |  pinv_2/gnd = 1
  EL_G3_sky130_fd_bd_sram__openram_dff/vdd = 1                                    |  pinv_3/vdd = 1
                                                                                  |
Net: gnd                                                                          |(no matching net)
  EL_G3_sky130_sram_2kbyte_1rw1r_32x512_8_pinv_2/gnd = 1                          |
  EL_G3_sky130_sram_2kbyte_1rw1r_32x512_8_pinv_3/gnd = 1                          |
  EL_G3_sky130_fd_bd_sram__openram_dff/gnd = 1                                    |
---------------------------------------------------------------------------------------------------------------------------------------------------------------------
```

Will also log a netgen issue.

RTimothyEdwards, 2023-04-11:
I don't think recent code changes have anything to do with this.  It is a very weird problem that can only happen in `ext2spice` because the .ext file for openram_dff does not even contain any references to the uppercase names.  It *does* have upper and lower case "CLK" and "clk" and those got reduced to just "clk" in the netlist.  So it appears that "VDD" and "GND" are being gratuitously added to the port list for no obvious reason.

RTimothyEdwards, 2023-04-11:
@d-m-bailey :  I think there is a quick solution here while I work on the underlying problem.  The superficial problem is that `extract no all` was used to prevent extraction of parasitics.  However, that also sets `extract no aliases` which is preventing the (non-port) labels from being output in an `equiv` statement.

There are actually a lot of things going wrong to make this error happen.  The "merge" statements are preferring the non-port name "VDD" to the name "vdd" when making the node merge in the parent cell.  OpenRAM is not writing either label "VDD" or "vdd" as a port to the DFF cell.  And "extract no all" is turning off alias name output in the .ext files.

I suggest for now replacing `extract no all` with `extract no all ; extract do aliases` in the extraction script.  I need to fix the problem where the `merge` connections are being made to the non-preferred name (which might be difficult since neither label is marked as a port, although there is at least supposed to be a strict typographical order to determine what is a "preferred" name).

d-m-bailey, 2023-04-11:
Great! I'll make the change to the extraction scripts and retry. Thanks for the suggestion.

d-m-bailey, 2023-04-11:
Adding `extract do aliases` remedied the connectivity issue. 

Getting size errors in the memory cell now... I check to see if it's netgen version related.