RTimothyEdwards/magic #234: Hierarchical extraction on pin/labels without drawing incomplete

d-m-bailey, 2023-04-15:
This occurs in slot-031 of mpw-8.

The reram array has a met2 strap to `SEL` in the base `cell_1t1r_lv`  reram cell. 
In neighboring placements of the `cell_1t1r_lv` cells in `subRA_2R2C`, `SEL<0:1>` are connected correctly.
```
.subckt NJ_YH_cell_1t1r_lv_0 SEL N P VSUBS
X0 N SEL a_65_65# VSUBS sky130_fd_pr__nfet_01v8 ad=0.4 pd=2.68 as=0.125 ps=1.17 w=0.89 l=0.15 M=2
X1 P a_65_65# sky130_fd_pr__reram_reram_cell area_ox=0.0254
.ends

.subckt NJ_YH_subRA_2R2C_0 N<0> N<1> SEL<0> P<1> SEL<1> VSUBS P<0>
XNJ_YH_cell_1t1r_lv_0_0 SEL<1> N<0> P<0> VSUBS NJ_YH_cell_1t1r_lv_0
XNJ_YH_cell_1t1r_lv_0_1 SEL<1> N<1> P<1> VSUBS NJ_YH_cell_1t1r_lv_0
XNJ_YH_cell_1t1r_lv_0_2 SEL<0> N<0> P<0> VSUBS NJ_YH_cell_1t1r_lv_0
XNJ_YH_cell_1t1r_lv_0_3 SEL<0> N<1> P<1> VSUBS NJ_YH_cell_1t1r_lv_0
.ends

.subckt NJ_YH_cell_1t1r_lv SEL N P VSUBS
X0 N SEL a_65_65# VSUBS sky130_fd_pr__nfet_01v8 ad=0.4 pd=2.68 as=0.125 ps=1.17 w=0.89 l=0.15 M=2
X1 P a_65_65# sky130_fd_pr__reram_reram_cell area_ox=0.0254
.ends

.subckt NJ_YH_subRA_2R2C N<0> N<1> SEL<1> SEL<0> P<1> VSUBS P<0>
XNJ_YH_cell_1t1r_lv_0_0 SEL<1> N<0> P<0> VSUBS NJ_YH_cell_1t1r_lv_0
XNJ_YH_cell_1t1r_lv_0_1 SEL<0> N<0> P<0> VSUBS NJ_YH_cell_1t1r_lv_0
XNJ_YH_cell_1t1r_lv_0 SEL<1> N<1> P<1> VSUBS NJ_YH_cell_1t1r_lv
XNJ_YH_cell_1t1r_lv_1 SEL<0> N<1> P<1> VSUBS NJ_YH_cell_1t1r_lv
.ends
```

However, in neighboring placements of `subRA_2R2C` in `subRA_2R16C`, the lower level met2 layer connectivity is broken.
```
.subckt NJ_YH_subRA_2R16C SEL<0> SEL<1> N<15> N<12> N<11> N<1> N<14> N<2> N<6> N<3>
+ N<4> N<5> N<13> N<10> N<7> N<8> N<9> P<12> P<11> P<2> P<10> P<3> P<4> P<8> P<5>
+ P<13> P<9> P<6> P<7> VSS P<1> NJ_YH_subRA_2R2C_1/SEL<1> NJ_YH_subRA_2R2C_1/SEL<0>
+ P<14> P<0> P<15> N<0>
XNJ_YH_subRA_2R2C_0_0[0] N<2> N<3> NJ_YH_subRA_2R2C_1/SEL<0> P<3> NJ_YH_subRA_2R2C_1/SEL<1>
+ VSS P<2> NJ_YH_subRA_2R2C_0
XNJ_YH_subRA_2R2C_0_0[1] N<4> N<5> NJ_YH_subRA_2R2C_0_0[1]/SEL<0> P<5> NJ_YH_subRA_2R2C_0_0[1]/SEL<1>
+ VSS P<4> NJ_YH_subRA_2R2C_0
XNJ_YH_subRA_2R2C_0_0[2] N<6> N<7> NJ_YH_subRA_2R2C_0_0[2]/SEL<0> P<7> NJ_YH_subRA_2R2C_0_0[2]/SEL<1>
+ VSS P<6> NJ_YH_subRA_2R2C_0
XNJ_YH_subRA_2R2C_0_0[3] N<8> N<9> NJ_YH_subRA_2R2C_0_0[3]/SEL<0> P<9> NJ_YH_subRA_2R2C_0_0[3]/SEL<1>
+ VSS P<8> NJ_YH_subRA_2R2C_0
XNJ_YH_subRA_2R2C_0_0[4] N<10> N<11> NJ_YH_subRA_2R2C_0_0[4]/SEL<0> P<11> NJ_YH_subRA_2R2C_0_0[4]/SEL<1>
+ VSS P<10> NJ_YH_subRA_2R2C_0
XNJ_YH_subRA_2R2C_0_0[5] N<12> N<13> SEL<0> P<13> SEL<1> VSS P<12> NJ_YH_subRA_2R2C_0
XNJ_YH_subRA_2R2C_1 N<1> N<0> NJ_YH_subRA_2R2C_1/SEL<1> NJ_YH_subRA_2R2C_1/SEL<0>
+ P<0> VSS P<1> NJ_YH_subRA_2R2C
XNJ_YH_subRA_2R2C_0 N<14> N<15> SEL<1> SEL<0> P<15> VSS P<14> NJ_YH_subRA_2R2C
.ends
```
The right 2 instances `XNJ_YH_subRA_2R2C_0_0[5]` and `XNJ_YH_subRA_2R2C_0 N<14>` are connected as expected, but the left 6 instances have unconnected hierarchical nets such as `NJ_YH_subRA_2R2C_1/SEL<0>`.

Bottom level
[image]
Middle level (*one level down)
[image]
Top level
[image]

Executing `./run_ext` in the attached tarball should recreate the problem.
[sel_connect.tgz](https://github.com/RTimothyEdwards/magic/files/11240006/sel_connect.tgz)

d-m-bailey, 2023-04-15:
Looks like the met2 wires have variable extensions. Does magic handle wires with variable extensions?
[image]

RTimothyEdwards, 2023-04-15:
Yes, the wire extension is being read in correctly.
This is the type of issue that I think "extract do labelcheck" was supposed to handle---the metal pin sits by itself but connects downward and across in the hierarchy.  I don't know whether it got broken recently due to all the changes in the extraction code, or if it's just a situation that was never being handled.  But I can investigate.  The layout is pretty simple, which makes it easier.

RTimothyEdwards, 2023-04-15:
I think that this is caused by the use of an array, and there may be something in the extraction code that doesn't anticipate the use of arrays.  My gut feeling is that if you replace the array of six devices in the middle with six individual instances, it will extract correctly.

RTimothyEdwards, 2023-04-15:
I also get "Number of subscripts doesn't match" errors when extracting, which is a bit weird as I don't recall having seen that before.  That suggests that the extraction failed because it failed to correctly count the number of instances in the array, which might be caused by something like the brackets in the cell name.