RTimothyEdwards/magic #236: Add `ANTENNAGATEAREA` for abstract lef views

kareefardi, 2023-04-19:
Magic is capable of writing a lef file with `ANTENNAGATEAREA` but only without `-hide` (non-abstract). It is beneficial to have it also in abstract lef view. See https://github.com/The-OpenROAD-Project/OpenLane/pull/1760 for more details

RTimothyEdwards, 2023-04-22:
There is nothing in the code that suggests that it should only write `ANTENNAGATE` when the `-hide` option is not selected.  Do you have a reproducible example?

kareefardi, 2023-04-26:
@RTimothyEdwards [reproducible.tar.gz](https://github.com/RTimothyEdwards/magic/files/11331289/reproducible.tar.gz)

RTimothyEdwards, 2023-04-26:
Must be that there is no search in the hidden area for diffusion and poly.  I'll take  a look at it.

RTimothyEdwards, 2023-04-26:
@kareefardi :  Fixed in version 8.3.395.

kareefardi, 2023-04-27:
Abstract lef has `ANTENNAGATEAREA` now.

closed 2023-04-27 by kareefardi