RTimothyEdwards/magic #239: magic extraction cannot discern nfet_03v3 from nfet_03v3_dw in GF180MCUC

spnadig, 2023-04-22:
@RTimothyEdwards  -  attached is a gds file with 2 kinds of inverters with nfets. One with DNWELL and the other without - which should correspond to nfet_03v3_dw and nfet_03v3 respectively. The extracted netlist (attached with extraction script used) however shows same devices (nfet03v3) for both cases. 
GF lists them to be seperate devices - as they must be mapped to different models -

[image]

[image]

[inv_dnwTest.zip](https://github.com/RTimothyEdwards/magic/files/11299997/inv_dnwTest.zip)

spnadig, 2023-04-22:
@proppy

proppy, 2023-04-22:
Looks like the spice deck also use `nmos_3p3` and not `nfet03v3` as the naming convention, see:
https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_pr/blob/main/models/ngspice/smbb000149.ngspice

@mithro can you link to the document that specify authoritatively the name of the primitive.

/cc @atokmabrains

atorkmabrains, 2023-04-22:
@proppy It's 'nfet_03v3' in here:

https://github.com/efabless/globalfoundries-pdk-libs-gf180mcu_fd_pr/tree/main/models/ngspice

Google repo is not up to date.

mithro, 2023-04-22:
https://bit.ly/gf180mcu-names

RTimothyEdwards, 2023-04-22:
The model name is the same whether or not the device is in deep nwell or not, so why even make the distinction?

spnadig, 2023-04-22:
Oh I was under the impression that models change with DNWELL given GF classifies them as different devices in their list?

RTimothyEdwards, 2023-04-22:
@spnadig :  The first table that you posted has a column for "Model Name" that indicates that the same model is used for both the device inside and outside of deep nwell.  I have not looked at it carefully;  maybe the device is just the underlying FET device?  There might be additional components in the subcircuits (i.e., does the deep nwell device have additional terminals and/or diode components representing the deep nwell and substrate connections?).

spnadig, 2023-04-22:
No there are no additional connections - it just has DNWELL layer and guard ring (which has 2 additional layers) around a typical nfet

RTimothyEdwards, 2023-04-22:
@spnadig :  I've used PDKs in the past where the deep nwell device models have additional connections representing the deep nwell and substrate in addition to the pwell and diffusion.  That makes them 5- and 6- terminal devices.  But in this case, I just looked at the GF SPICE models;  there are no subcircuit models for the devices in deep nwell.

closed 2023-04-24 by spnadig