RTimothyEdwards/magic #240: Increasing of spacing of metal1 and nwell layer in GDS generated File

VanshikaTanwar, 2023-04-26:
As, I have generated the GDS  file of Analog part of 3 bit flash ADC using ALIGN Tool so, when I am opening the layout in magic tool then there are 3 DRC errors there, I have tried removing those DRC's eroors but not able to solve it !! 

Hence, I need your help in solving this DRC errors.
Here, are the all file which I have attached below

[image]

[image]

[image]

![WhatsApp Image 2023-04-26 at 07 32 41(1)](https://user-images.githubusercontent.com/90523478/2

[image]

[image]

RTimothyEdwards, 2023-04-26:
This is an ALIGN issue, not a magic issue.  If you are using tools to automate the process of layout, then those tools need to be DRC aware.  If the tools don't automatically handle all DRC issues, then you need to understand how to do layout manually so that you can correct the problems.  Both issues are simple enough, although from a bird's-eye view, it looks like ALIGN has chosen to flip part of the circuit vertically, which causes the wells to be placed corner to corner, which is a generally bad floorplan.  The floorplan may even prevent the nwell spacing error from being correctable in this case.

VanshikaTanwar, 2023-04-27:
Ok , so ,now how should I remove these DRC error sir? As ,sir I tried removing these DRC's error I am failed to do it .