RTimothyEdwards/magic #242: Import of GDS failed.

JakobRat, 2023-05-09:
Hi,
I imported a GDS file into magic but unfortunately the poly-layer wasn't imported correctly.

[image]

You can see the difference by the selected instances in the right part of the figure. 

[GDS_Error.zip](https://github.com/RTimothyEdwards/magic/files/11429208/GDS_Error.zip)

RTimothyEdwards, 2023-05-09:
@JakobRat :  You probably have a PDK from the brief period where the polysilicon import was screwed up (sadly, the worst errors posted for the shortest time are the ones that manage to get propagated everywhere---Murphy's Law, I suppose).  The issue was introduced in PDK (open_pdks) version 1.0.406 and corrected in PDK version 1.0.407 (the next day!).

JakobRat, 2023-05-10:
@RTimothyEdwards  Thank you! That's exactly the case.

closed 2023-05-10 by JakobRat