RTimothyEdwards/magic #243: Extraction chooses non-port label

d-m-bailey, 2023-05-17:
Using magic `8.3.398`

When there are multiple labels on a net, magic sometimes chooses the non-port label which results in LVS port mismatch.

The net* signals in `serv_1` port list are internal signals that are shorted to ports.
`serv_1.v`
```
 assign io_oeb[0] = net566;
 assign io_oeb[1] = net567;
 assign io_oeb[2] = net568;
 assign io_oeb[3] = net569;
 assign io_oeb[4] = net570;
 assign io_out[2] = net571;
 assign io_out[3] = net572;
 assign io_out[4] = net573;
```
`serv_1.ext`
```
equiv "io_oeb[0]" "net566"
equiv "io_oeb[1]" "net567"
equiv "io_oeb[2]" "net568"
equiv "io_oeb[3]" "net569"
equiv "io_oeb[4]" "net570"
```
`user_project_wrapper.gds.spice`
```
.subckt serv_1 net568 io_in[3] net569 net567 net573 net571 net570 io_out[1] io_out[0]
+ io_in[0] io_in[4] io_in[1] io_in[2] net566 net572 vdd vss
```
This results in mismatched ports at th `serv_1` level
```
Circuit 1: serv_1                                                       |Circuit 2: serv_1
------------------------------------------------------------------------|------------------------------------------------------------------------
net571                                                                  |io_oeb[3] **Mismatch**
```
and after altering ports to match,  at the top level
```
 serv_1 u_serv_1 (.vdd(vdd),
...
    .io_oeb({_08_,
    _07_,
    _06_,
    io_oeb[18],
    io_oeb[17]}),
```
Top level pin list.
```
io_oeb[18]                                                              |(no matching pin)
io_oeb[17]                                                              |(no matching pin)
```

Explicitly flattening `serv_1` results in a match. 
Ports are expected to have precedence over internal node text.

To reproduce,  
```
export PDK=gf180mcuC
tar xzf test-gflvs.tgz
cd test-gflvs
./run_test
```
Not a minimal case, but only takes about 10 minutes.
`lvs.noflatten.report` will contain the mismatched results without flattening.
`lvs.flatten.report` will contain the matched results with flattening (the size errors are netgen issue).
[test-gflvs.tgz](https://github.com/RTimothyEdwards/magic/files/11503024/test-gflvs.tgz)

RTimothyEdwards, 2023-06-19:
@d-m-bailey :  Revisiting this layout and netgen result, I discovered that the example has an incorrect netlist for the standard cell library antenna cell.  It does not match what is produced by open_pdks for the library.  The netlist has the diode device connected to VDD and VSS.  However, the antenna cell layout has the diode device connected to VPW and VNW, not VDD and VSS, so that is an actual error.  Probably the attempt to match VDD and VSS pins on both sides is what is causing the error.

I also don't understand how the layout-extracted netlist is created, because it is missing the VNW and VPW ports for the standard cells, and transistors don't have source/drain area and perimeter output.

d-m-bailey, 2023-06-19:
@RTimothyEdwards Just to make sure we're on the same page. The sample data I sent was for gf180 not sky130. I think gf180 uses VDD and VSS instead of VPW and VNW.

RTimothyEdwards, 2023-06-20:
@d-m-bailey :  Yes, GF.  The original GF standard cells do not have VPW and VNW, but they do have the same setup with each standard cell having well and substrate disconnected from power and ground, and expected to be connected through a tap cell.  So to make it all correct, I had to add the port names VPW and VNW to all the standard cell views, which is done by a script during the PDK build.

RTimothyEdwards, 2023-06-20:
(Actually the original CDL netlists from GF had the VPW and VNW ports named, which is where I got the nomenclature from.  I think that was the _only_ format that explicitly named the ports.)

RTimothyEdwards, 2023-06-20:
I assert that from tests that I've done on the most recent code, the initially reported issue of non-port labels being selected over port labels has been fixed.  Other issues with running LVS on the contents of the test case tarball are unrelated.  I believe that this issue can be closed.

d-m-bailey, 2023-06-20:
@RTimothyEdwards I reviewed the gf180 LVS results that I have for the taped out data. It may be that the gds was created before the change to the pdk to add `VPW` and `VNW`. For the standard cells, the psubstrate connection was extracted as VSUB and the nwell connection as an auto-assigned node, eg. `w_n86_352#`. 
As for the missing area and perimeter mosfet parameters, I used `version 1.0.414-0-gaf34855`, but other than that, I don't know why they wouldn't be extracted either. Maybe I'm not setting the style correctly.

Thanks for fixing this issue. I'll let you know if I see any other problems.

RTimothyEdwards, 2023-06-21:
@d-m-bailey :  The tarball you posted above (test-gflvs.tgz) is self-contained with a tech file `version 1.0.359-0-g35c7265`.  However, I did the GDS read, extraction, and LVS using the most recent versions of everything (which is a PDK version of `1.0.393-3-g815863e` and magic version 8.3.404).  But the GDS file itself has VPW and VNW labels (on the correct layers) in the standard cells in the layout, so I'm not sure why they extracted without the labels for _any_ version.  "Extraction chooses non-port label" was an issue involving a port label for a non-port label.  Choosing a non-label over a label is a completely separate issue (and one which I cannot reproduce with this GDS).

d-m-bailey, 2023-06-23:
@RTimothyEdwards Here's a sky130 example (mpw-8 `slot-001`). Takes about 10 minutes to run. 

The `user_proj_example` level is `JM_LO_wb_pio`. 
Here's the subckt header (notice `net160`)
```
.subckt JM_LO_wb_pio io_oeb[35] io_in[27] net160 io_oeb[23] io_out[28] io_out[21]
```
From the ext file,
```
node "net160" 0 0 252569 277185 v0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
equiv "net160" "io_oeb[37]"
...
merge "wb_pio_160/LO" "net160" 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
```

```
tar xzf test_ext.tgz
cd test_ext
./run_ext
```
Maybe it has something to with the `[test_ext.tgz](https://github.com/RTimothyEdwards/magic/files/11848088/test_ext.tgz)` statement.

RTimothyEdwards, 2023-06-23:
@d-m-bailey :  Can you provide either a minimal test case or a test case that does not use oasis?  I cannot run the converter script because I do not know what the "pya" package is.  It does not get installed by "pip3 install pya".

d-m-bailey, 2023-06-23:
@RTimothyEdwards I tried using the [gds in the repo](https://foss-eda-tools.googlesource.com/third_party/shuttle/sky130/mpw-008/slot-001/+/refs/heads/main/gds/user_project_wrapper.gds.gz), but that worked fine. It's just over 25MB so I can't upload here.
The data that I use that has the error is extracted from the final caravel, but it is over 25MB also.

Here's the `pya/__init__.py` file.
```
$ cat /usr/local/lib/python3.7/dist-packages/pya/__init__.py
# import all packages from klayout, such as klayout.db and klayout.tl
# WARNING: doing it manually until it becomes impractical
# TODO: We need a specification document explaining what should go into pya

from klayout.db import *  # noqa
from klayout.lib import *  # noqa
from klayout.tl import *  # noqa
from klayout.rdb import *  # noqa
```

So that looks like it's part of the `klayout` installation. If you don't have klayout installed, can you try it in docker?

d-m-bailey, 2023-06-24:
@RTimothyEdwards Ok, so I just split the tar file.
```
cat test_ext.001.tar.gz test_ext.002.tar.gz | tar xzf -
cd test_ext
./run_ext
```
[test_ext.002.tar.gz](https://github.com/RTimothyEdwards/magic/files/11856008/test_ext.002.tar.gz)
[test_ext.001.tar.gz](https://github.com/RTimothyEdwards/magic/files/11856009/test_ext.001.tar.gz)

RTimothyEdwards, 2023-06-27:
@d-m-bailey :  Perhaps I needed to be more clear about wanting a _minimal_ test case.  Anyway, I was able to remove 99.9% of the layout and all but three pins and it continued to exhibit the problem.  I eventually tracked it down;  I think prior to creating the "def read" option that preserves net names, I had not encountered a layout with a port and a non-port label on the same net.  The synthesis tools apparently name nets separately from ports, so with the implementation (and use) of "def read" with net names, plus the switch to "extract do aliases" being default, suddenly made this a common thing.  There was, in fact, no code to keep labels sorted between ports and non-ports when labeling a net region during extraction, so any label could end up being the "preferred" name of a node.  That was actually easy to fix once I understood the problem.

RTimothyEdwards, 2023-06-27:
A fix has been applied in magic version 8.3.408.

d-m-bailey, 2023-06-27:
Excellent!