RTimothyEdwards/magic #247: DRC clean designs drop contacts on GDS export

d-m-bailey, 2023-05-25:
On one of the [mpw-3 analog designs](https://foss-eda-tools.googlesource.com/third_party/shuttle/sky130/mpw-003/slot-016/), the tapeout data is [missing licon1](https://github.com/d-m-bailey/extra_be_checks/discussions/27#discussioncomment-5905914) that is visible in magic. 
[image]
Note: cell has been rotated.
[image]

The gds feedback shows 1816 errors
```
      8 feedback add "CIF error in cell Class_D_post, layer CONT: no room for contacts in area!" pale
   1800 feedback add "CIF error in cell Class_D_post, layer MCON: no room for contacts in area!" pale
      8 feedback add "CIF error in cell Class_D_post, layer NPC: no room for contacts in area!" pale
```
but there were no drc errors indicating that there would be a problem. Magic `8.3.400`

Maybe this should be a fatal error that produces no gds by default.

RTimothyEdwards, 2023-05-25:
@d-m-bailey :  It's problematic to force that to be a fatal error because there are situations where it crops up on complicated polygon geometry---some areas are too small to fit contacts but it is not an issue for the overall region.  But the "+" shape above has caused the issue in a way that is not detected as a DRC error.  There is probably a way to detect that kind of issue;  it's a pathological case.  Maybe a count of vias generated in any single connected area, such that if the count is zero then a fatal error occurs.  Either than, or in the above case, the via generation routine needs to be able to deal with that pathological case.  At any rate, that case is one that I wasn't aware could happen.

d-m-bailey, 2023-05-25:
@RTimothyEdwards Thanks for checking. 
So _most_ of the time, the errors are related to something like this
[image]
where some vias are generated and some aren't. In this case too, the user might want to be know that only one via was created where they were expecting 3.

A quick and dirty solution might be to add a option `cif allowerrors [yes|no]` with a default of `no`.
At the end of `cif write <file>`, 
if errors are detected and `allowerrors` is `no`, 
- delete `<file>`
- display `<file> not created due to cif write errors. Use "cif allowerrors yes" to create <file> in spite of errors.`

if errors are detected and `allowerrors` is `yes`
- display `Warning: <file> created in spite of errors.`

RTimothyEdwards, 2023-05-26:
@d-m-bailey :  Exactly.  The cut filling algorithm will start with a contact tile and then try to expand it in all directions to find the largest area available to place cuts in.  What makes the example that you posted up top so pathological is that the "+" shape can be expanded two ways, either up-down or side-to-side, so both ways would need to be checked to discover that a cut can only fit in one of those expansions.

RTimothyEdwards, 2023-05-26:
Your example directly above is _not_ such a case, and is not a situation in which the user should expect that there will be three contacts---the area of the contact is only slightly bigger than a via.  The main issue here is that any contact area that is not flagging a DRC error should have at least one cut, and the pathological case is one instance (and possibly the only instance?) in which no DRC error occurs but no cut is placed.