RTimothyEdwards/magic #248: Top connection of m5-m4 capacitors extracted as floating

d-m-bailey, 2023-06-02:
Version `8.3.392`

There are 7 m4-m5 mimcaps in the design 1 is 6.88x6.89, 3 are 2x2, and 3 are 1x1.

The 1x1 capacitors extract with a floating top net.
```
X11 c2_7893_4697# a_6172_2126# sky130_fd_pr__cap_mim_m3_2 l=1 w=1
X12 c2_531_9153# VSS sky130_fd_pr__cap_mim_m3_2 l=1 w=1
X15 c2_6142_4445# a_6361_537# sky130_fd_pr__cap_mim_m3_2 l=1 w=1
```

Expect the top connections to be extracted as shown in the layout.
```
tar xzf lvs.tgz
gunzip RX_Frontend.spice.gz
cd lvs
./run_lvs RX_Frontend ../RX_Frontend.spice ../user_analog_project_wrapper-lvs.gds.gz
```
Extracted netlist is in  `RX_Frontend/ext/RX_Frontend.gds.spice`

[lvs.tgz](https://github.com/RTimothyEdwards/magic/files/11635783/lvs.tgz)
[user_analog_project_wrapper-lvs.gds.gz](https://github.com/RTimothyEdwards/magic/files/11635784/user_analog_project_wrapper-lvs.gds.gz)
[RX_Frontend.spice.gz](https://github.com/RTimothyEdwards/magic/files/11635796/RX_Frontend.spice.gz)

RTimothyEdwards, 2023-06-03:
I have to run some numbers, but it looks like my rule deck for magic might need to adjust the size and rules of the VIA4 layer over MiM cap capm2.  Magic is assuming that the m4 overlap of via4 rule applies for MiM caps, so if the MiM cap isn't at least 1.18um x 1.18um, the contact won't fit and may be lost on GDS read-in.

RTimothyEdwards, 2023-06-03:
It is made more difficult by having the wrong type of MiM cap and rules in the sky130 open PDK published documentation.

RTimothyEdwards, 2023-06-07:
This is specifically a sky130 PDK issue, not a magic issue, and should be raised in the open_pdks repository, not here.  For the record---The magic tech file for sky130 needs to have the capm2 and capm2c rules adjusted so that a minimum size capm2 can be drawn and will be legal by the DRC rules in magic.   That will keep the area from being considered too small for the contact which will prevent the contact from being deleted on GDS read-in.

RTimothyEdwards, 2023-06-09:
@d-m-bailey :  FYI, the problem is that the layout you looked at is following a set of (unfortunate) rules that was in place in the s130 PDKs between some unknown revision and revision 305 (which is pretty recent).  It looks like SkyWater inadvertently used an incorrect value for required metal4 surround of via4, taking the value from the rules for some incompatible variant of s8.  Although the original s8 rules and the sky130 rules, and the s130 rev 305+ rules all agree, there was a period of time (maybe a year or two) in which the wrong value was in the s130 PDK.  So I stand by the values that I have in the sky130 PDK.  The via4 shapes in the layout GDS you're looking at are not surrounded by enough metal 4, so they are too small to fit a via4 contact cut.

RTimothyEdwards, 2023-06-09:
However, it would be a nice feature if magic would issue an alert when certain critical layers get axed from the input GDS.