RTimothyEdwards/magic #250: Parasitics extraction of generic resistor results in floating node

simonwaid, 2023-06-19:
When extracting parasitics adding a generic resistors results in a floating node. Please find attached a sample file. [test.zip](https://github.com/RTimothyEdwards/magic/files/11792855/test.zip). The file contains two ports connected via a resistor on metal 5. One of the ports also connects to the substrate. One would expect no floating nodes when extracting the circuit. 

Extraction was done using the following command:
`select top cell; extract do local; extract all; ext2sim labels on; ext2sim; extresist tolerance 10; extresist; ext2spice lvs; ext2spice cthresh 0; ext2spice extresist on; ext2spice`

simonwaid, 2023-06-20:
Agreed, the example is wrong. I have the issue in a larger circuit and tried  to create a minimum example. Please find attached a new version of the example: [test1.zip](https://github.com/RTimothyEdwards/magic/files/11804314/test1.zip) The extracted circuit contains the capacitor C0 between `m5_1800_0# `and P4.`m5_1800_0#` is a floating node.

RTimothyEdwards, 2023-06-20:
@simonwaid :  This may not be the optimal solution, and I don't know if it works in all cases, but it worked for this one;  which is to use the option `ext2spice resistor tee on`.  That extracts the resistor as a T-model, splitting the resistor in half and giving the parasitic capacitance of the device to the midpoint node.

The underlying issue is that the resistor device itself is treated as a floating node, so any capacitance to it ends up as a floating capacitance.

The error is that there is not a default "resistor pi" model that redistributes the capacitance of the device itself (or coupling capacitance to the device) to the terminal nodes.  I will look into a better solution for it, although the "resistor tee" option should be sufficient for you to use as a workaround.