RTimothyEdwards/magic #259: Empty tkcon About box; can't read "RCS": no such variable

ryandesign, 2023-07-21:
When I launch `magic`, a tkcon v2.3 console opens. If I choose "About" from its Help menu, an Application Error window appears saying:

> Error: can't read "RCS": no such variable

[image]

If I click "Details >>", it says:

```
can't read "RCS": no such variable
can't read "RCS": no such variable
    while executing
"$w.text insert 1.0 "About tkcon v$PRIV(version)" title  "\n\nCopyright 1995-2001 Jeffrey Hobbs, $PRIV(email) \nRelease Info: v$PRIV(version), CVS v$RC..."
    (procedure "::tkcon::About" line 21)
    invoked from within
"::tkcon::About"
    invoked from within
".#mbar.#mbar#help invoke active"
    ("uplevel" body line 1)
    invoked from within
"uplevel #0 [list $w invoke active]"
    (procedure "tk::MenuInvoke" line 50)
    invoked from within
"tk::MenuInvoke .#mbar.#mbar#help 1"
    (command bound to event)
```

If I click "OK", an About tkcon v2.3 window appears that contains nothing:

[image]

macOS 12 ships with /usr/bin/tkcon. If I launch that and open its About box, it works properly:

[image]

So the 2.3 version of tkcon you're bundling with magic appears to be at least three releases older than one from 2010. So, pretty old. The [latest version of tkcon](https://github.com/wjoye/tkcon) released last year appears to be 2.7.11.

Could you update your bundled tkcon to a newer version?

Or is it even necessary for you to bundle tkcon at all? Would it be reasonable to instead use an already-installed tkcon, requiring your users to install it themselves first? That way they, or the OS or package manager they use, can be responsible for keeping it updated, freeing you of the responsibility.

RTimothyEdwards, 2023-07-21:
@ryandesign :  I made a few changes to the tkcon.tcl script to work with my tools;  the official version will not work out-of-the-box with magic.  Somewhere I have written down what changes I made.  I have updated tkcon.tcl in the past without too much trouble, but those handful of changes prevents me from just using a pre-installed tkcon.

RTimothyEdwards, 2023-07-21:
I think the RCS was supposed to be replaced by running "make" in the github clone, and I just copied the tkcon.tcl file without doing "make" and ended up with an invalid script without noticing it.

RTimothyEdwards, 2023-07-25:
@ryandesign :  I discovered that other of my opencircuitdesign.com software projects like IRSIM use the same tkcon.tcl (modified), but it looks like the RCS string in the version in magic got truncated somehow.  Since both copies of tkcon.tcl came from the same original version, I just copied back the RCS string and it works now.  This is currently the simpler option than updating tkcon.tcl to the latest, since there are a handful of places it gets modified, and I need to apply the same modifications to any new version.

Fixed in magic version 8.3.417.

closed 2023-08-01 by ryandesign