RTimothyEdwards/magic #260: Extraction issue

kareefardi, 2023-07-30:
In the attached layout, magic generates a connection between GND pin of the macro inside the layout and the top level VGND pins of the layout in the spice extracted file. By inspecting the layout itself, I can't find this connection. 
This is the layout [blackbox_test.gds.gz](https://github.com/RTimothyEdwards/magic/files/12209022/blackbox_test.gds.gz). This is the script used for extraction: 
```
gds read blackbox_test.gds
load blackbox_test
select top cell
expand
extract do local
extract no all
extract do aliases
extract unique
extract all
ext2spice lvs
ext2spice -o blackbox_test.spice blackbox_test.ext
```

d-m-bailey, 2023-07-30:
The connection may be through the psubstrate if both the top level and macro have psub taps. psubstrate is a connecting layer in magic. [extra_be_checks](https://github.com/d-m-bailey/extra_be_checks/tree/oeb-check) has a soft connection check that detects these connections.

kareefardi, 2023-07-30:
@d-m-bailey It appears that was the problem. Thanks.

closed 2023-07-30 by kareefardi