RTimothyEdwards/magic #261: Compile error: implicit-function-declaration

NickOveracker, 2023-08-08:
Encountered when attempting to **make** in Mint:

```
gcc -c -DHAVE_CONFIG_H   -I. -I. -DRL_LIBRARY_VERSION='"4.3"' -g -m64 -fPIC -Werror=implicit-function-declaration   display.c
display.c: In function ‘rl_redisplay’:
display.c:605:22: error: implicit declaration of function ‘wcwidth’ [-Werror=implicit-function-declaration]
  605 |               temp = wcwidth (wc);
      |                      ^~~~~~~

```

RTimothyEdwards, 2023-08-08:
Okay, I have fixed this error in the source.  However, this error comes from compiling "readline", which means that you are *not* building the Tcl/Tk version of magic, which is very much *not* recommended.  You will end up with a version of magic that is crippled relative to what everyone else is using.  Make sure you have installed the proper development packages for Tcl and Tk, then do "make clean" and start over with "configure".

NickOveracker, 2023-08-09:
I see. Thanks!

hamidb80, 2023-08-14:
Hey, I have `tclsh` and `wish` installed, but I don't know how to find tcl header files ti build with magic.

Could you please help me?

hamidb80, 2023-08-15:
I have `tcl8.6` installed in my Ubuntu 22 machine, but `configure` cannot detect it:

```bash
./configure --with-tcl=/usr/include/tcl.8.6 --with-tk=/usr/include/tcl8.6/
```

output:
```
...
Tcl/Tk:       no

  Without Tcl/Tk, you cannot run the enhanced version of magic
  with the toolbar and menus, and a number of other functions
  are disabled.  If you did not specifically disable Tcl/Tk on
  the configure command line, then getting this message means
  that you do not have Tcl/Tk headers and or libraries installed,
  or they are not in a standard path.  Try using configure options
  --with-tcl=<DIR> and --with-tk=<DIR>.

-----------------------------------------------------------
Use 'make' to compile and 'make install' to install.
```

RTimothyEdwards, 2023-08-15:
@hamidb80 :  The "development" packages for Tcl/Tk are different from the "distribution" packages and contain headers that are needed for compiling other applications with Tcl/Tk.  You should not need to specify `--with-tcl` and `--with-tk` on the configuration line, but you do need to do `sudo apt install tcl-dev tk-dev`.

After that, do `make clean` before running `configure` without options.