RTimothyEdwards/magic #262: Netlists created with ext2spice short resistor should be equivalent to those created without.

d-m-bailey, 2023-08-18:
With `ext2spice short resistor`, MPW-3 slot-019 extracts with 2 shorts in `AW_RD_analog_mux`.
```
R0 EN1 ENB1 0.000000
R1 VGND_uq0 VGND 0.000000
```
`EN1` to `ENB1` is a true error, but `VGND_uq0` and `VGND` are connected through metal2 in subcells.

Flattened view
[image]

Top only
[image]

Comparing the 2 netlists with netgen gives
```
Subcircuit summary:
Circuit 1: AW_RD_analog_mux                |Circuit 2: AW_RD_analog_mux
-------------------------------------------|-------------------------------------------
AW_RD_analog_switch (2)                    |AW_RD_analog_switch (2)
sky130_fd_pr__pfet_g5v0d10v5 (7)           |sky130_fd_pr__pfet_g5v0d10v5 (7)
sky130_fd_pr__nfet_03v3_nvt (7)            |sky130_fd_pr__nfet_03v3_nvt (7)
Number of devices: 16                      |Number of devices: 16
Number of nets: 12                         |Number of nets: 12
---------------------------------------------------------------------------------------
Netlists match uniquely with port errors.

Subcircuit pins:
Circuit 1: AW_RD_analog_mux                |Circuit 2: AW_RD_analog_mux
-------------------------------------------|-------------------------------------------
SELECT                                     |SELECT
EN                                         |EN
EN0                                        |EN0
ENB0                                       |ENB0
ENB1                                       |EN1 **Mismatch**
EN1                                        |EN1
VPWR                                       |VPWR
VGND                                       |VGND
VGND_uq0                                   |(no matching pin)
VOUT                                       |VOUT
VIN1                                       |VIN1
VIN0                                       |VIN0
---------------------------------------------------------------------------------------
Port number 11 greater than number of ports 10
Port number 12 greater than number of ports 10
Cell pin lists for AW_RD_analog_mux and AW_RD_analog_mux altered to match.
Device classes AW_RD_analog_mux and AW_RD_analog_mux are equivalent.
  Flattening non-matched subcircuits AW_RD_analog_mux AW_RD_analog_mux
```

The `Port number ?? greater than number of ports 10` may indicate an inconsistency with the internal netlist because the top results give the following disconnected ports along with many mismatched nets. Note, this is the same layout being extracted with 2 different options - one would expect a match (or near match).
```
Cell AW_user_analog_project_wrapper (1) disconnected node: /AW_RD_analog_mux_0//AW_RD_analog_switch_0/VGND
Cell AW_user_analog_project_wrapper (1) disconnected node: /AW_RD_analog_mux_0//AW_RD_analog_switch_1/VGND
Cell AW_user_analog_project_wrapper (1) disconnected node: /AW_RD_analog_mux_0//AW_RD_analog_switch_1/ENB
Cell AW_user_analog_project_wrapper (1) disconnected node: /AW_RD_analog_mux_0/sky130_fd_pr__pfet_g5v0d10v5:1/3
Cell AW_user_analog_project_wrapper (1) disconnected node: /AW_RD_analog_mux_0/sky130_fd_pr__nfet_03v3_nvt:2/4
Cell AW_user_analog_project_wrapper (1) disconnected node: /AW_RD_analog_mux_0/sky130_fd_pr__nfet_03v3_nvt:3/3
Cell AW_user_analog_project_wrapper (1) disconnected node: /AW_RD_analog_mux_0/sky130_fd_pr__nfet_03v3_nvt:3/4
Cell AW_user_analog_project_wrapper (1) disconnected node: /AW_RD_analog_mux_0/sky130_fd_pr__pfet_g5v0d10v5:5/1
Cell AW_user_analog_project_wrapper (1) disconnected node: /AW_RD_analog_mux_0/sky130_fd_pr__nfet_03v3_nvt:6/2
Cell AW_user_analog_project_wrapper (1) disconnected node: /AW_RD_analog_mux_0/sky130_fd_pr__nfet_03v3_nvt:6/3
Cell AW_user_analog_project_wrapper (1) disconnected node: /AW_RD_analog_mux_0/sky130_fd_pr__nfet_03v3_nvt:6/4
Cell AW_user_analog_project_wrapper (1) disconnected node: /AW_RD_analog_mux_0/sky130_fd_pr__nfet_03v3_nvt:7/1
Cell AW_user_analog_project_wrapper (1) disconnected node: /AW_RD_analog_mux_0/sky130_fd_pr__nfet_03v3_nvt:7/4
Cell AW_user_analog_project_wrapper (1) disconnected node: /AW_RD_analog_mux_0/sky130_fd_pr__pfet_g5v0d10v5:10/2
Cell AW_user_analog_project_wrapper (1) disconnected node: /AW_RD_analog_mux_0/sky130_fd_pr__nfet_03v3_nvt:11/1
Cell AW_user_analog_project_wrapper (1) disconnected node: /AW_RD_analog_mux_0/sky130_fd_pr__nfet_03v3_nvt:11/4
Cell AW_user_analog_project_wrapper (1) disconnected node: /AW_RD_analog_mux_0/sky130_fd_pr__nfet_03v3_nvt:12/3
Cell AW_user_analog_project_wrapper (1) disconnected node: /AW_RD_analog_mux_0/sky130_fd_pr__nfet_03v3_nvt:12/4
Cell AW_user_analog_project_wrapper (1) disconnected node: /AW_RD_analog_mux_0/sky130_fd_pr__nfet_03v3_nvt:13/3
Cell AW_user_analog_project_wrapper (1) disconnected node: /AW_RD_analog_mux_0/sky130_fd_pr__nfet_03v3_nvt:13/4
```

To duplicate 
```
tar xzf test.tgz
cd test
./run_test
```
[test.tgz](https://github.com/RTimothyEdwards/magic/files/12379646/test.tgz)

This will extract with the option `ext2spice short resistor` to `ext.analog` and without the option to `ext.digital`.
The results are then compared.

Maybe this is a case where the ports should not be altered to match.

RTimothyEdwards, 2023-09-04:
This is a confounding example and I will need to spend more time looking into it.

As far as I can tell, the issue with "VGND_uq0" is superficial and has nothing to do with the other errors.  Maybe.
In addition to the "VGND_uq0" problem, I was confused as to why "VPWR_uq0" is created in exactly the same way but does not appear in the netlist.

I corrected the "VGND_uq0" in a way that I don't really like, by identifying during "extract all" that it is shorted to "VGND" and therefore not unique, and reverting the unique node label back to the original.  But this is wrong for many reasons.  I don't like the entire "extract unique" method to begin with.  There should be no method that alters labels inside the layout.  The labels should be handled "behind the scenes" during extraction, leaving the layout itself unchanged.  "extract unique" should set a flag for use during "extract".  That will take a bit of work to do properly.

However, assuming that the code change I made is doing the correct thing by reverting "VGND_uq0" back to "VGND" in the layout, this does not change the "Port number. . . greater than number of ports" error or any of the other problems that show up during LVS.  So I haven't gotten to the bottom of the basic problem with "extract short".

RTimothyEdwards, 2023-09-04:
@d-m-bailey :  See my comments where you left an issue in the issue tracker for Magic.  I looked at the two netlists being created in "analog" and "digital" and they only differ in the zero-valued resistor in "analog".  The zero-valued resistor connects to a port that is itself not connected to anything, so these netlists should be equivalent.  I made a correction to netgen to fix the problem, which is unrelated to the "extract unique" problem in Magic discussed above.