RTimothyEdwards/magic #265: env: python3: No such file or directory

ryandesign, 2023-08-29:
magic 8.3.425 fails to build on older systems that don't have a `python3` executable, like macOS 10.14 and earlier:

```
--- making header file database/database.h
./scripts/makedbh database/database.h.in database/database.h
env: python3: No such file or directory
make: *** [database/database.h] Error 127
```

This problem is new in 8.3.425 and was caused by 02e16b8bce8f9d19fef32932a4742e797025cc19.

I am already telling the configure script where my version 3.x python executable is located, e.g.:

```
./configure ac_cv_path_PYTHON3=/opt/local/bin/python3.11
```

but this makedbh script is ignoring that.

ryandesign, 2023-08-29:
I see that scripts/preproc.py also begins with `#!/usr/bin/env python3` but that's not new in 8.3.425 and has not been a problem. I guess normal builds don't run preproc.py and so don't run into this problem there. Presumably the python 3 that the configure script found (or was told to use) should be used in both preproc.py and makedbh.

ryandesign, 2023-08-29:
I remember now: I fixed this for preproc.py in b4d635a3abae58b73b53efc3625b91ce2c282555.

ryandesign, 2023-08-29:
It's not as easy a fix for makedbh because the Makefile from which makedbh is called is not processed by the configure script.

RTimothyEdwards, 2023-08-29:
@donn :  Your code is already causing problems.  I'm going to revert your pull request if you don't fix it.

donn, 2023-08-29:
See #266 - @ryandesign, your review would be appreciate as well

closed 2023-09-03 by ryandesign