RTimothyEdwards/magic #267: Error summary appears to be incomplete.

d-m-bailey, 2023-09-13:
Reading the tech file yields
```
PDK_ROOT/sky130A/libs.tech/magic/sky130A.tech: line 19: section version:
        Error:  Magic version 8.3.411 is required by this techfile, but this version of magic is 8.3.348.
Input style sky130(): scaleFactor=2, multiplier=2
PDK_ROOTsky130A/libs.tech/magic/sky130A.tech: line 5715: section extract:
        Unrecognized layer (type) name "a1>0.99"
PDK_ROOT/sky130A/libs.tech/magic/sky130A.tech: line 5715: section extract:
...
The following sections of PDK_ROOT/sky130A/libs.tech/magic/sky130A.tech contained errors:
    version
```
Since errors occur in both the `version` section and the `extract` section, how about listing `extract` in the final summary.

RTimothyEdwards, 2023-09-14:
This appears to be the policy about tech file errors, from ExtTechLine():
```
 * Results:
 *      Returns TRUE normally, or FALSE if the line from the
 *      technology file is so malformed that Magic should abort.
 *      Currently, we always return TRUE.
```
From that, I gather that the final message implies "The following sections . . . contained _fatal_ errors".  I think it would be necessary to add additonal tracking of non-fatal errors.  But I think the general idea is that it's more important to know which section was the one that caused magic to exit immediately.

d-m-bailey, 2023-09-14:
There were no fatal errors. magic ran to completion when just merging data.

RTimothyEdwards, 2023-09-14:
@d-m-bailey :  Oh.  Not satisfying the tech file requirement is supposed to bring magic to an immediate halt.  I will have to check that.

RTimothyEdwards, 2023-09-14:
I think this needs some work. Clearly not satisfying the version requirement does not cause magic to halt, which I now realize is a source of issues with open_pdks.  I had been assuming that I could force an update to magic by causing it not to run if the version did not meet the requirement.  The only workable solution as I see it is to use two separate return values, one for non-fatal errors and one for fatal errors.  Report all sections with all types of errors, and then quit magic if there was a fatal error.

RTimothyEdwards, 2023-09-14:
@d-m-bailey:  I have decided that exiting on a fatal error is easy enough to do in the startup script---If there was a fatal error, then the tech file will not get loaded, and the "tech name" command will return something other than the expected tech name, which is easy to check and handle.  That means there is no particular need to treat tech file errors as "fatal" or "non-fatal", and all errors should cause the corresponding routine to return FALSE so that all sections with errors are properly reported (which goes back to your initial statement).  So the proper solution is just to make sure that all tech file read routines correctly set their own return value to FALSE whenever they encounter an error.