RTimothyEdwards/magic #271: Segmentation fault when running *watch active demo

gkamendje, 2023-10-17:
I get a segmentation fault when running `*watch active demo` or `*watch metal1 demo`. 
I am running Magic `8.3.434-1-g656b5db` on Ubuntu 22.04.3/WSL. I saw the same behavior on CentOs7 with  `8.3.435`. 
To reproduce: 
```
/path_to_magic/magic/8.3.434-1-g656b5db/bin/magic maint2a
```
Then either type the command `*watch active demo` or `*watch metal1 demo`

RTimothyEdwards, 2023-10-17:
Duly noted, and I can reproduce the crash condition.  I just fixed it in the code, and will push it along with some other code edits I have planned for today.

Meanwhile, you can work around the issue by forcing the layout to be editable:

```
magic maint2a
cellname writeable maint2a true
select top cell
edit
*watch active demo
```

gkamendje, 2023-10-17:
Copying maint2a to a local writable directory also works. Thanks for the hint.

closed 2023-10-22 by gkamendje