RTimothyEdwards/magic #272: Load log clarification request.

d-m-bailey, 2023-10-19:
`Magic 8.3 revision 431`
When loading cells, the cell reference listed in the `mag` file is displayed as below.
```
Cell gpio_defaults_block_087 read from path /home/user/gf180mcu/mpw-1/caravel_user_project/mag
```
If the cell can not be found at that reference, the following messages are displayed.
```
Warning:  Parent cell lists instance of "mprj_io_buffer" at bad file path /home/user/gf180mcu/mpw-1/caravel_user_project/mag/mprj_io_buffer.mag.
The cell exists in the search paths at mprj_io_buffer.mag.
The discovered version will be used.
```
It would be helpful to display the path of the discovered version to verify what data is actually used.

RTimothyEdwards, 2023-10-19:
I believe that this is true, that `The cell exists in the search paths at mprj_io_buffer.mag` means that it was found in the current directory.

d-m-bailey, 2023-10-24:
You are correct. The full message does indeed state that the cell is being read from the current directory.
```
Warning:  Parent cell lists instance of "mprj_io_buffer" at bad file path /home/user/gf180mcu/mpw-1/caravel_user_project/mag/mprj_io_buffer.mag.
The cell exists in the search paths at mprj_io_buffer.mag.
The discovered version will be used.
Cell mprj_io_buffer read from current working directory
```

closed 2023-10-24 by d-m-bailey