RTimothyEdwards/magic #276: does not build if python3 less-than 3.6 (e.g. python3.5): SyntaxError: invalid syntax

sto6, 2023-10-28:
The usage of f-string requires python3.6. If the found python3 is 3.5 the build fails.
My example happens to be on a centos-7, where python3.6 is available (installed) but /usr/bin/python3 is python3.5 due to the requirements of other system on the host.

Is it true: python3.6 (apparently) is now a requirement of the BUILD; but only the BUILD and still no python is required at runtime??

(See also #265 @donn.)

```
--- making header file database/database.h
./scripts/makedbh database/database.h.in database/database.h
  File "./scripts/makedbh", line 16
    f"Bad regular expression in {sys.argv[0]}: Cannot generate database/database.h!",
                                                                                   ^
SyntaxError: invalid syntax
make[1]: *** [Makefile:48: database/database.h] Error 1
```

RTimothyEdwards, 2023-10-28:
@sto6 :  Donn did that.

RTimothyEdwards, 2023-10-28:
@sto6 :  I just modified it so that it will be compatible with any Python-3 revision.

donn, 2023-10-28:
Sorry about that.

That said, every Python version before 3.8 is EOL: https://devguide.python.org/versions/

sto6, 2023-10-28:
Confirmed, thanks. 
Please describe python3 dependency in INSTALL and: http://opencircuitdesign.com/magic/install.html

donn, 2023-10-28:
While you're there, remove 'csh'/'tcsh' (no longer needed)