RTimothyEdwards/magic #277: Layout Window not opening in macOS(Sonoma)

pingu-73, 2023-10-31:
I followed the instructions in the macOS install file. But when i run magic command in terminal it only opens **magic console window** and not the **layout window**.

[image]

d-m-bailey, 2023-10-31:
@pingu-73 Do you have `XQuartz` running?

pingu-73, 2023-10-31:
@d-m-bailey i've XQuartz installed. Do i need to run it before opening magic?

pingu-73, 2023-10-31:
on running **./scripts/configure_mac** summary is the following
[image]

RTimothyEdwards, 2023-10-31:
@pingu-73 :  Huh.  Everything looks fine right up to the point where it says "using NULL graphics".  What was the command line you used to start magic?  Do you get a different result if you use `magic -d OGL` or `magic -d XR`?

d-m-bailey, 2023-10-31:
@pingu-73 Yes, XQuartz needs to be active.

pingu-73, 2023-10-31:
@d-m-bailey even after activating XQuartz i'm getting same result

[image]

d-m-bailey, 2023-10-31:
Any thing at the start of the console log file?

pingu-73, 2023-10-31:
@d-m-bailey 
[image]

d-m-bailey, 2023-10-31:
As @RTimothyEdwards says, it shouldn't say `Using NULL graphics device`. Did you try `magic -d OGL` or `magic -d XR`

d-m-bailey, 2023-10-31:
Did you get any error messages when following the  `INSTALL_MacOS.md` instructions?

pingu-73, 2023-10-31:
@RTimothyEdwards @d-m-bailey  using `magic -d OGL` gives segmentation fault but `magic -d XR` doesn't shows any error. Also i didn't get any error message while following `INSTALL_MacOS.md`
[image]

[image]

RTimothyEdwards, 2023-10-31:
Type `export DISPLAY=:0` on the command line before running magic and see if that causes a window to pop up.  I worked through this issue with somebody else two months ago and they had the same issue that magic would start but not generate a window, and it didn't generate an error, either.

pingu-73, 2023-11-01:
@RTimothyEdwards now i'm not even able to open magic console
[image]

pingu-73, 2023-11-02:
@RTimothyEdwards i've solved this issue and made few changes in macOS install file can you review those changes and merge them.
[PR #278 ](https://github.com/RTimothyEdwards/magic/pull/278)

closed 2023-11-02 by pingu-73