RTimothyEdwards/magic #279: .magicrc ignored

jacobnrohan, 2023-11-14:
I'm running qflow using the suggested demo source file [map9v3.v](http://opencircuitdesign.com/qflow/example/map9v3.v). The flow runs successfully through the Migration step using the osu018 technology. Placement and Routing steps show their respective animations. **However, when I click "Edit Layout" in qflow, magic opens with an empty layout and no technology layers**, and qflow reports the following:
```
Diagnostic:   Creating qflow.magicrc
Diagnostic:   Found a tech load command in .magicrc
Diagnostic:   tech load not trusted
Diagnostic:   Inserting own tech load
Magic version is 8.3.443
```
For clarity, after clicking "Edit Layout", qflow.magicrc is generated:

[image]

**However, nothing appears in magic:**
[image]
<details>
  <summary>
    magic log
  </summary>
<pre><code>
loading history file ... 26 events added
Use openwrapper to create a new GUI-based layout window
Use closewrapper to remove a new GUI-based layout window

Magic 8.3 revision 443 - Compiled on Thu Nov  2 09:18:22 AM CDT 2023.
Starting magic under Tcl interpreter
Using Tk console window
Processing system .magicrc file
Could not find file 'SCN4M_SUBM.20.tech' in any of these directories:
         . $CAD_ROOT/magic/sys $CAD_ROOT/magic/sys/current
4 Magic internal units = 1 Lambda
New windows will not have a title caption.
New windows will not have scroll bars.
New windows will not have a border.
Root cell box:
           width x height  (   llx,  lly  ), (   urx,  ury  )

microns:   0.000 x 0.000   ( 0.000,  0.000), ( 0.000,  0.000)
lambda:     0.00 x 0.00    (  0.00,  0.00 ), (  0.00,  0.00 )
internal:      0 x 0       (     0,  0    ), (     0,  0    )
Reading LEF data from file /usr/local/share/qflow/tech/osu035/osu035_stdcells.lef.
This action cannot be undone.
LEF read, Line 104 (Error): Don't know how to parse layer "metal1"
LEF read, Line 104 (Error): Try adding this name to the LEF techfile section
LEF read, Line 105 (Error): No layer defined for RECT.
LEF read, Line 106 (Error): Don't know how to parse layer "via1"
LEF read, Line 106 (Error): Try adding this name to the LEF techfile section
LEF read, Line 107 (Error): No layer defined for RECT.
LEF read, Line 108 (Error): Don't know how to parse layer "metal2"
LEF read, Line 108 (Error): Try adding this name to the LEF techfile section
LEF read, Line 109 (Error): No layer defined for RECT.
LEF read, Line 113 (Error): Don't know how to parse layer "metal2"
LEF read, Line 113 (Error): Try adding this name to the LEF techfile section
LEF read, Line 114 (Error): No layer defined for RECT.
LEF read, Line 115 (Error): Don't know how to parse layer "via2"
LEF read, Line 115 (Error): Try adding this name to the LEF techfile section
LEF read, Line 116 (Error): No layer defined for RECT.
LEF read, Line 117 (Error): Don't know how to parse layer "metal3"
LEF read, Line 117 (Error): Try adding this name to the LEF techfile section
LEF read, Line 118 (Error): No layer defined for RECT.
LEF read, Line 122 (Error): Don't know how to parse layer "metal3"
LEF read, Line 122 (Error): Try adding this name to the LEF techfile section
LEF read, Line 123 (Error): No layer defined for RECT.
LEF read, Line 124 (Error): Don't know how to parse layer "via3"
LEF read, Line 124 (Error): Try adding this name to the LEF techfile section
LEF read, Line 125 (Error): No layer defined for RECT.
LEF read, Line 126 (Error): Don't know how to parse layer "metal4"
LEF read, Line 126 (Error): Try adding this name to the LEF techfile section
LEF read, Line 127 (Error): No layer defined for RECT.
LEF read, Line 132 (Error): Don't know how to parse layer "metal1"
LEF read, Line 132 (Error): Try adding this name to the LEF techfile section
[... 111 lines cut ...]
Contact size value ignored (using GDS generation rules).
Contact size value ignored (using GDS generation rules).
The following types are not handled by extraction and will be treated as non-electrical types:
    n_field_implant p_field_implant nselect pselect glass filln filla fillb xp m1p m2p m3p m4p comment bb 
4 Magic internal units = 1 Lambda
New windows will not have a title caption.
New windows will not have scroll bars.
New windows will not have a border.
Repainting console in magic layout window colors
Root cell box:
           width x height  (   llx,  lly  ), (   urx,  ury  )

microns:   0.000 x 0.000   ( 0.000,  0.000), ( 0.000,  0.000)
lambda:     0.00 x 0.00    (  0.00,  0.00 ), (  0.00,  0.00 )
internal:      0 x 0       (     0,  0    ), (     0,  0    )
Reading LEF data from file /usr/local/share/qflow/tech/osu035/osu035_stdcells.lef.
This action cannot be undone.
LEF read: Processed 3179 lines.
Cell map9v3 read from current working directory
Scaled magic input cell map9v3 geometry by factor of 2
Processing timestamp mismatches.
Timestamp mismatches found in these cells: INVX8, XNOR2X1, INVX4, INVX2, AND2X2, OAI22X1, XOR2X1, NOR3X1, INVX1, NOR2X1, NAND3X1, NAND2X1, OAI21X1, AOI21X1, AOI22X1, DFFSR, BUFX2, CLKBUF1, FILL.
Using technology "scmos", version 2001a
</code></pre>
</details>

RTimothyEdwards, 2023-11-15:
Looking at the python script that generates the error message, it looks like it is expecting
```
tech load /usr/local/share/qflow/tech/osu035/SCN4M_SUBM.20
```
instead of
```
path sys +/usr/local/share/qflow/tech/osu035
tech load SCN4M_SUBM.20
```
Since it's a python script parsing a Tcl file, it's not prepared to understand the entire script.  I can put in behavior to check for the `path sys` command, though, which should solve the immediate problem.

closed 2023-11-15 by RTimothyEdwards

RTimothyEdwards, 2023-11-15:
As far as I could tell, there were two separate errors here, as the ".tech" extension is often left off the name of the tech file in the "tech load" command, but the qflow script was not adding the extension before searching if the file exists.  I have now added code to properly handle the file extension, and also to properly handle the "path sys" lines.

Ultimately this comes from an over-simplification of the magic startup script by the python code in qflow, and the fact that I switched to using "path sys" in recent years because I now mostly use technologies from the `open_pdks` installer, which puts technology files in paths with awkwardly long names like `/usr/local/share/pdk/sky130A/libs.tech/magic/sky130A.tech` which I found much cleaner to present in two separate lines.

RTimothyEdwards, 2023-11-15:
The fix is in qflow version 1.4.101 (pushed to opencircuitdesign.com this morning and which will be mirrored to github by tomorrow).

jacobnrohan, 2023-11-17:
Installed qflow version 1.4.101 and can confirm my issue is resolved. **Thank you!**