RTimothyEdwards/magic #289: Segmentation fault

wulffern, 2024-02-03:
I experience a segmentation fault when I try to netlist a layout. 

I run 

```tcl
set VDD AVDD
set GND AVSS
set SUB BULKN
load ../design/SUN_PLL_SKY130NM/SUN_PLL.mag
#writeall force
extract all
ext2spice lvs
ext2spice -o lvs/SUN_PLL.spi
quit
```

On my [design](https://github.com/wulffern/sun_pll_sky130nm), and magic segfaults. I think it happens because somehow magic is trying to check the size of a empty cell, although I'm not sure, but the segfault happens in bpBins.c at

```C
int ew = GEO_WIDTH(&e->e_rect);
int eh = GEO_HEIGHT(&e->e_rect);
```

I've fixed it with 

https://github.com/wulffern/magic/commit/68b89ab774014c8e58459adebd0c4da8c1fb8bd7

RTimothyEdwards, 2024-02-05:
I would want to be _very_ careful with changes to the code around the `bplane` code.   I will see if I can reproduce the problem from your repo.

RTimothyEdwards, 2024-02-05:
@wulffern :  I can't duplicate the issue because the design repository is incomplete;  there are multiple cell layouts missing.

wulffern, 2024-02-05:
Thanks for looking into it. Found one more piece of info.

If I open the cell with `magic ../design/SUN_PLL_SKY130NM/SUN_PLL.mag &`
and immediately do `extract all`, then it crashes

However, if I open, draw a box around the cells, expand all cells with hotkey x, wait for the DRC to finish, then `extract all`, it works fine.

RTimothyEdwards, 2024-02-05:
And the even weirder thing that I found was that if I add the equivalent commands that make it work into the script, that is, `select top cell` and `expand` after the `load` command, then it _still_ segfaults when running in batch.

RTimothyEdwards, 2024-02-05:
And another weird thing is that I can run the script directly on SUN_PLL_LPF.mag and it works fine, even though that was the cell it was working on when it crashed.
I may have to put this one on the back burner for a bit.  I don't think I can solve it without running a debugger on the working and non-working methods and figuring out where they diverge.  Your solution is a workaround but the real source of the problem is somewhere else.

wulffern, 2024-02-11:
Found another workaround.

 If I replace `extract all` with just `extract`, then first time it crashes, but second, and later times, it runs fine. 

If I delete all *.ext files, and run, then it crashes the first time.

RTimothyEdwards, 2024-02-11:
"extract" as opposed to "extract all" is an incremental extraction, so after the first run, it will not attempt to extract any cell for which a .ext exists and is more recent than the layout.

RTimothyEdwards, 2024-02-11:
Actually, I am finding that the SUN_PLL cell does not extract correctly even when magic doesn't crash (i.e., when subcells are expanded before extracting).  The output is incomplete, and a warning is issued during "ext2spice".

RTimothyEdwards, 2024-02-19:
@wulffern :  Finally had time to look into this, and I ended up patching the code to circumvent the issue in much the same way you did, which is pretty much what I suggested originally except that I wanted to make sure I understood the code surrounding the issue.  Thank you for providing the example!

closed 2024-02-22 by wulffern