RTimothyEdwards/magic #291: Strange extraction of cell arrays.

d-m-bailey, 2024-02-07:
magic 8.3.456

sky130 mpw-7 slot-020 reram project has a 2x2 array of reram cells.

The ext file for `KM_SC_1T1R_2x2` has
```
merge "KM_SC_1T1R_0[0:1,1:1]/bl" "bl" 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
merge "bl" "KM_SC_1T1R_0[0:1,0:0]/bl"
merge "KM_SC_1T1R_0[1:1,0:1]/sl" "sl" 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
merge "sl" "KM_SC_1T1R_0[0:0,0:1]/sl"
merge "KM_SC_1T1R_0[1:1,0:1]/VGND" "VGND" 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
merge "VGND" "KM_SC_1T1R_0[0:0,0:1]/VGND"
merge "KM_SC_1T1R_0[0:1,1:1]/wl" "wl" 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
merge "wl" "KM_SC_1T1R_0[0:1,0:0]/wl"
```
and the log file contains warnings for these 8 lines.
```
KM_SC_1T1R_2x2.ext, line 16: Number of subscripts doesn't match
```

The expectation is that there would be no warnings.

Here's the gds. 
[KM_user_analog_project_wrapper.lvs.gds.gz](https://github.com/RTimothyEdwards/magic/files/14197090/KM_user_analog_project_wrapper.lvs.gds.gz)

d-m-bailey, 2025-10-17:
@pei108 I continue to see a problem, however, it may not be with the bus formatted instance names. 

Sometimes the connectivity is not extracted correctly. (differs if arrays are used or not).

I'm currently trying a workaround that un-arrays in the gds before extracting, but the resulting layout is not exactly the same.

RTimothyEdwards, 2025-10-17:
There are a few unrelated problems here, so please keep them distinct.

There is a trivial error in which magic reports "Number of subscripts doesn't match".  This message started popping up when I started having to support brackets in cell names, a result of this use being common in xschem when declaring a multiple set of instances connected to bus inputs and outputs.  Magic had been using brackets exclusively for arrays, so I had to fix up the code to disambiguate brackets in cell names vs. brackets indicating arrays.  This appears to work correctly everywhere except for the spurious error message.  Because it's only a messaging issue and does not appear to reflect a real error, fixing it has been low priority.

There is a separate issue where magic will fail to correctly extract an array.  This appears to have something to do with having the array span multiple extraction tiled areas.  A net name on one end of the array fails to get matched to the name of the same net on the other side of the array.  Or something like that.  It is high priority but also a very difficult problem to debug.

RTimothyEdwards, 2025-10-17:
@pei108 :  Magic _does_ support extracting designs with cell arrays.  Occasionally there are errors as mentioned in my post above.  If you have an example of an array which fails to extract correctly, please post it so I can have another example to look at.

RTimothyEdwards, 2025-10-17:
It's easy enough to add `mask-hints CONT` in front of any point in the cifoutput section that outputs the CONT layer.

I have been wanting to look at a test case big enough to give the `gds maskhints` function a thorough vetting.  It's nice to know that it more or less works as advertised.  Known obvious failure points are if the tech file doesn't define mask hints for a layer in the output, like CONT, or if the layer in the GDS is _smaller_ than the layer output by magic, which is a lot harder to deal with (it's always easy enough just to add more stuff).  Non-manhattan geometry is not supported (but implants, which are the usual target of mask hints, rarely ever have non-manhattan geometry).

What other differences did you find?  My ultimate goal is to be able to trust the automatic mask-hints handling to produce GDS that can be trusted to match the original.