RTimothyEdwards/magic #292: DEF Write Error

thud06, 2024-02-22:
Im currently using magic version 8.3. When I try to write a def file using def -write i get the error: "free() : invalid pointer" then : "Aborted (core dumped)

RTimothyEdwards, 2024-02-22:
Version 8.3 point what?

I'm happy to track it down but I will need a reproducible result.

thud06, 2024-02-22:
Woops sorry! It is Version 8.1 revision 213 when I type the version command. It is crashing when I try to write a def on version 7.5.241 on as well so I assume I might just be doing something wrong. My process is:

1. Create a cell using SCN3ME_SUBM.30.TECH27
2. save
3. writeall
4. save
5. extract all
6. def write fileName

It is at this point the magic console says: 

Generating DEF output filename.def for cell cellName:
Diagnostic:  Write DEF header for cell cellName
Diagnostic:  Finding all components of cell cellName
Diagnostic:  Finding all nets in cell cellName
(This can take a while!)

It then crashes with the "free() : invalid pointer" then : "Aborted (core dumped) errors on my linux console.

RTimothyEdwards, 2024-02-22:
Version 8.1.213 is six years out of date.  The current version of magic is 8.3.462.

thud06, 2024-02-22:
I went ahead and updated to 8.3.462 and am getting the same error. The magic console is saying "freeMagic called with NULL argument."

RTimothyEdwards, 2024-02-22:
@thud06 :  Can you please post the tech file?  I did the steps above for a layout from qflow with the osu050 technology.  Where did your tech file come from?  It might simply be too old (although I try hard to retain backwards compatibility);  the .tech27 format has not been used for decades.  I have an updated tech file in the qflow repository which is the same technology but with a few critical differences such as a substrate specification, the lack of which might be what's causing the crash condition.  I can go figure out if I need to fix the backwards-compatibility, while you can try the modified tech file and see if that's a good workaround for you (https://github.com/RTimothyEdwards/qflow) (file is tech/osu050/SCN3ME_SUbM.30.tech) (note the file extension ".tech" instead of ".tech27").

RTimothyEdwards, 2024-02-23:
@thud06 :  Yes, but can you still please post the original tech file that you used?  I have (supposedly) the same file (from public MOSIS ftp archives that no longer exist), but magic doesn't segfault when I use it.  I do get a strange behavior where pins and metal layers get confused with each other.  But it didn't crash the program.

thud06, 2024-02-23:
Sure! GitHub doesnt let me upload this file type so here it is copied/pasted:

tech
 scmos
end

version
 version 2002a
 description "SCMOS: Submit as technology.lambda: SCN3ME_SUBM.30 [to process: AMIc5]"
end

planes
  well,w
  implant,i
  select,s
  cap,c
  active,a
  metal1,m1
  metal2,m2
  metal3,m3
  oxide,ox
  comment
  xp
  contact
  via1,v1
  via2,v2
  fill
end

types
  well  nwell,nw
  well  pwell,pw
  implant  n_field_implant,nfi
  implant  p_field_implant,pfi
  select  nselect,ns
  select  pselect,ps
  cap  electrode,poly2,el,p2
  cap  electrodecontact,poly2contact,poly2c,p2c,elc
  cap  p2m12contact,p2m12c
  cap  electrodecap,ecap,poly2cap,p2cap,pcap
  contact  genericpoly2contact,gc2
  active  ntransistor,nfet
  active  ptransistor,pfet
  active  diffusion,diff
  active  transistor,fet
  active  ndiffusion,ndif,green
  active  pdiffusion,pdif,brown
  active  ndcontact,ndc
  active  pdcontact,pdc
  active  psubstratepdiff,pohmicdiff,pod,ppdiff,ppd,psd
  active  nsubstratendiff,nohmicdiff,nod,nndiff,nnd,nsd
  active  psubstratepcontact,pohmiccontact,poc,pwcontact,pwc,psc
  active  nsubstratencontact,nohmiccontact,noc,nwcontact,nwc,nsc
  active  polysilicon,red,poly,p
  active  polycontact,pcontact,polycut,pc
  contact  genericcontact,gcontact,gc
  metal1  metal1,m1,blue
  metal1  pseudo_rmetal1,prm1
  metal1  rmetal1,rm1
  metal1  fillm1,fm1
  metal1  m2contact,m2cut,m2c,via1,v1
  metal1  pm12contact,pm12c
  metal1  pdm12contact,pdm12c
  metal1  psm12contact,psm12c,pom12c,pwm12c
  metal1  ndm12contact,ndm12c
  metal1  nsm12contact,nsm12c,nom12c,nwm12c
  metal2  metal2,m2,purple
  metal2  pseudo_rmetal2,prm2
  metal2  rmetal2,rm2
  metal2  fillm2,fm2
[... 8454 lines cut ...]
 4141 2222 0000 2222
 ndiff,rnd,ndc/a,ndm12c/a,nfet \
 0808 1010 2020 4040 \
 8080 4141 2222 1414 \
 0808 1010 2020 4040 \
 8080 4141 2222 1414
 pc/a,pm12c/a,ndc/a,ndm12c/a,pdc/a,pdm12c/a,psc/a,psm12c/a,nsc/a,nsm12c/a,gc,gc,gc X

style gremlin
 pfet 9
 nfet 10
 gv1 11
 pwell 15
 nwell 16
 poly,fp,rp,pc/a,pm12c/a,nfet,pfet 19
 pc/a,pm12c/a,ndc/a,ndm12c/a,pdc/a,pdm12c/a,psc/a,psm12c/a,nsc/a,nsm12c/a,gc,gc,gc 22
 pad,glass 23
 nsd,nsc/a,nsm12c/a 24
 gv1 28
 pdiff,rpd,pdc/a,pdm12c/a,pfet 29
 psd,psc/a,psm12c/a 30
 ndiff,rnd,ndc/a,ndm12c/a,nfet 31
 pc/a,pm12c/a,ndc/a,ndm12c/a,pdc/a,pdm12c/a,psc/a,psm12c/a,nsc/a,nsm12c/a,gc,gc,gc,gv1 X

end

RTimothyEdwards, 2024-02-23:
Actually the thing to do is to tarball it into a .tgz file and then post it, which github will allow.  But this works.

RTimothyEdwards, 2024-02-23:
I'm glad the modified tech file works for you, because even with the file you posted I cannot duplicate the crash condition.  So I will have to shelve the problem.  Let me know if anything else crops up.