RTimothyEdwards/magic #293: Hierarchy under abstract cells is extracted and netlisted.

d-m-bailey, 2024-02-28:
In order to speed up extraction, known macros are sometimes extracted as black-boxes by setting `LEFview` property to true.

This results in a netlist that has the macro in question as a block box, but the subcells of the macro are extracted and written to the netlist anyway.

The desired outcome is that any cells that are only subcells of an abstracted cell are not extracted or netlisted.

RTimothyEdwards, 2024-02-28:
@d-m-bailey :  So. . .  can you provide a simple test case?

d-m-bailey, 2024-02-28:
Once my open source workstation comes back online, I'll make one.

d-m-bailey, 2024-03-16:
@RTimothyEdwards I've implemented a solution in the extraction script, so this is not a critical issue.