RTimothyEdwards/magic #294: make font size adjustable

mattvenn, 2024-03-24:
I and other people struggle with the tiny font size in Magic. Some way to adjust the font size, even at compile time would be appreciated. Even better would be via .magicrc or a setting in a menu

RTimothyEdwards, 2024-03-24:
I assume you are referring to the Tk text for menus and such?
Create a `.magicrc` in your home directory and add these lines:
```
set Opts(scale) 2.0
set Opts(toolscale) 1.0
```

isilbert, 2024-03-28:
> I assume you are referring to the Tk text for menus and such?
> 
> Create a `.magicrc` in your home directory and add these lines:
> 
> ```
> 
> set Opts(scale) 2.0
> 
> set Opts(toolscale) 1.0
> 
> ```
> 
> 

How would the Tkconsole shell text size be changed then? Currently the only option to change the size of the shell text appears to be to call magic with the -noconsole flag and adjust your terminal's text size manually (unless I am mistaken?). Is there something like this available?

RTimothyEdwards, 2024-03-29:
Oh, that probably does not change the console text size.
Magic has no control over that, because it is controlled from Tk and so is a property of the Wish application.
It should be possible to set X defaults for Wish in a .Xdefaults file, but it's something that I would need to look into, because I'm not even sure if the defaults are supposed to be set for "tkcon", "wish", or what.