RTimothyEdwards/magic #297: Devices from array not imported

EpSilicon, 2024-03-31:
Hello,
Transistor arrays, taken from a netlist created in Xschem, are not imported into Magic (XM6).
Thansk and regards,
Erwann

Magic version: 8.3.466
Xschem version: v3.4.5
SKY130 from open_pdks: 1.0.470 (Installed with Volare)

** sch_path: /home/erwann/design_worksapce/sky130A/HelloWorld/xschem/opamp.sch
.subckt opamp vdd out inp inn ibiasn vss
*.PININFO inn:I inp:I vdd:B vss:B ibiasn:B out:O
XM1 pbias inn tail vss sky130_fd_pr__nfet_01v8 L=1 W=40 nf=1 m=1
XM2 outz inp tail vss sky130_fd_pr__nfet_01v8 L=1 W=40 nf=1 m=1
XM3 pbias pbias vdd vdd sky130_fd_pr__pfet_01v8 L=0.5 W=20 nf=1 m=1
XM4 outz pbias vdd vdd sky130_fd_pr__pfet_01v8 L=0.5 W=20 nf=1 m=1
XM5 out outz vdd vdd sky130_fd_pr__pfet_01v8 L=0.5 W=20 nf=1 m=1
XM6[1] tail ibiasn vss vss sky130_fd_pr__nfet_01v8 L=1 W=40 nf=1 m=1
XM6[0] tail ibiasn vss vss sky130_fd_pr__nfet_01v8 L=1 W=40 nf=1 m=1
XM7 out ibiasn vss vss sky130_fd_pr__nfet_01v8 L=1 W=40 nf=1 m=1
XM8 ibiasn ibiasn vss vss sky130_fd_pr__nfet_01v8 L=1 W=40 nf=1 m=1
.ends
.end

RTimothyEdwards, 2024-03-31:
The brackets are probably causing an issue.  There may be an option in xschem to avoid putting brackets in the netlist output, or you can remove them manually, as a workaround.  I will fix it when I get the chance.  If I don't respond within a couple of weeks, please send me a reminder.

RTimothyEdwards, 2024-04-01:
@EpSilicon :  Yes, it was the brackets.  I have fixed the problem in magic version 8.3.467.

EpSilicon, 2024-04-02:
I can confirm that it works, thank you!

RTimothyEdwards, 2024-04-02:
Good to know.  Closing this issue.

closed 2024-04-02 by RTimothyEdwards