RTimothyEdwards/magic #299: Magic extraction appears to be stuck in a loop.

d-m-bailey, 2024-04-06:
magic version `8.3.465`

public design from 2311 slot-009

flatglobbing cells and creating 2 subcut regions. The details are in the `extract.tcl` file in [debug-2311-090.tgz](https://github.com/RTimothyEdwards/magic/files/14893688/debug-2311-090.tgz).

```
tar xzf debug-2311-090.tgz
cd debug-2311-090
magic -dnull -noc -rcfile magicrc extract.tcl
```

Normally, it should finish for a minute or two, but runs for hours without completing.

RTimothyEdwards, 2024-04-08:
For me, this example ran for about 24 seconds and completed without any issue.

d-m-bailey, 2024-04-09:
@RTimothyEdwards thanks for checking. It may have something to do with the changes I'm proposing for #296 .

What version are you using?