RTimothyEdwards/magic #300: LEF obstruction for nwell seems incomplete

smunaut, 2024-04-07:
So this is originally from a .mag

If you compare the generated GDS:
[image]

And then the generated LEF:
[image]

One of the large N-Well obstruction is missing from the LEF.

(Those all comes from PCell subcells in the original MAG. On the top it's two PCell overlapping their guardring).

RTimothyEdwards, 2024-04-07:
I'm not sure why it's missing the nwell shape.  I don't see a prBoundary layer;  if it doesn't exist, then magic will take the extents of geometry as the bounding box, and should include everything inside that bounding box in the LEF file.

smunaut, 2024-04-07:
There is a prBoundary, it definitely englobes all the shapes shown.

This is the design btw : https://github.com/TinyTapeout/tt_asw_1v8/blob/main/mag/tt_asw_1v8.mag

RTimothyEdwards, 2024-04-07:
Thanks for the link to the layout.  That's clearly wrong, and I'll need to investigate.

RTimothyEdwards, 2024-04-08:
@smunaut :  I found out that "lef write" has an inconsistent policy about MASTERSLICE layers.  The nwell that got found was part of the routine checking for stuff outside the prBoundary, which includes MASTERSLICE layers.  But the routine that fills in the center area when using the "-hide" option does not include MASTERSLICE layers.  After thinking it over, I decided to allow the center area to be filled with nwell.  That might be a bad solution, though, because it implies that there is nwell in areas of the cell where there isn't any.  So I have provisionally implemented that solution, but I am still thinking about it.