RTimothyEdwards/magic #304: Parasitic intermittent crashes, reproducibility of extract files

thesourcerer8, 2024-05-02:
There is a problem in the parasitic extraction of magic, sometimes it gives an error, sometimes it works, on the same file:
[image]
This is the INV.mag file which contains a simple inverter which allows to reproduce the issue:
[INV.mag.txt](https://github.com/RTimothyEdwards/magic/files/15186505/INV.mag.txt)

These are the commands used:
```
extract all
ext2sim labels on
ext2sim
extresist tolerance 10
extresist
```
And this is the whole tool where it is used: https://github.com/thesourcerer8/StdCellLib/blob/master/Tools/perl/parasitics.pl

One idea for a potential reason might be the ordering in the generated .ext file, which seems to be different:
[image]

One important thing for me is reproducibility, so if you see a way to make the ordering in the extract files reproducible by sorting the output, that would be great, (unless that creates unbearable CPU or memory load)

Or perhaps we should create an "extract reproducible" option so that it can be switched off by default, and reproducibility enforced only on demand?

RTimothyEdwards, 2024-05-04:
I've been aware of this issue but this is the first reasonably small test case I've seen demonstrating it.

FYI, there are still various subtle bugs in the extresist code.  They should have nothing to do with the random order of the output, which is due entirely to the fact that the output is generated by walking a hash table, and the hash table contents are keyed by memory location, so order is dependent on the memory allocation algorithm.  It is of course possible to sort the hash table contents before output;  there would be a memory overhead associated with it, and I would probably want to make it an option that can be turned on or off, but it's not hard to do.

RTimothyEdwards, 2024-05-05:
Magic version 8.3.478 fixes the error condition (and also a crash condition that got uncovered by fixing the error condition).  The order of capacitor lines in the .ext file is not considered.

thesourcerer8, 2024-05-05:
Great! I cannot reproduce the error anymore with 8.3.478 :+1: 
So I think we should close this bug, and have a seperate issue for the reproducible ordering of the lines?

RTimothyEdwards, 2026-11-05:
I had left this issue open because I had not works on the output reproducibility issue.  Since there is now a recent issue (#551) which flags the same issue, and since the other part of this issue was solved long ago, I'm going to close this and work on #551 instead.

closed 2026-11-05 by RTimothyEdwards