RTimothyEdwards/magic #305: erasing a corner makes labels fly to space

thesourcerer8, 2024-05-02:
I am not quite sure whether this is intentional behavior or a bug: If I erase a small corner of a piece of metal which has a label associated, the label suddenly flies to space and isn't attached to the metal anymore, even thought the remaining metal would be large enough to still have the label attached.

RTimothyEdwards, 2024-05-04:
This is intentional behavior, although "large enough" in this context means "completely covers the area defined by the label box".

A way to avoid having the label get reassigned to "space" is the command `setlabel sticky true`.  However, for some processes (like Sky130), it is considered a DRC error to have a label that is not completely covered by the material that it is supposed to be connected to.

thesourcerer8, 2024-05-04:
Hmm, to be honest I would prefer the label to shrink in the same way the covered material shrinks, or to resize the label to the largest rectangle coverable by the remaining area. Seen a different way: The label should stick at it's center to the material it covers, and then flood-fill to the material again. If the center isn't there anymore, I wouldn't mind it flying off to space.

RTimothyEdwards, 2024-05-04:
That's a nice implementation idea.  I'll look into it.

RTimothyEdwards, 2024-05-05:
Magic version 8.3.478 now implements this behavior.  I could make it a bit smarter by checking what is the largest remaining rectangular area under the label;  the current implementation just clips the label area back by the amount cut out, in order (left, right, top, bottom).  Regardless, the behavior of adjusting the label size while keeping it on its attached layer instead of reassigning it to "space" seems like a better choice.