RTimothyEdwards/magic #306: magic filename.mag shows only boring window since 8.7.472 with tcl enabled

mtasaka, 2024-05-03:
Hello.

When magic is compiled with configure option:

```
export WISH_EXE=/usr/bin/wish
configure \
	--with-tcl=/usr/lib64 \
	--with-tk=/usr/lib64 \
	--with-tcllibs=/usr/lib64} \
	--with-tklibs=/usr/lib64 \
```
since magic 8.3.476, loading example mag file e.g. `$ magic ./scmos/examples/ccd/ccd.mag` shows only boring window, compared with magic 8.3.471 as attached.

With magic 8.3.476:
[image]

With magic 8.3.471:
[image]

Note that 8.3.476 shows:
```
Cell ccd has technology "scmos", but current technology is "minimum"
Use command "tech load" if you want to switch technologies, or use
"cellname delete ccd" and "load ccd -force" to force the cell to load as technology minimum
```

This seems related with this change: https://github.com/RTimothyEdwards/magic/commit/82d64aa4b19fa8ec7e1f6b7237b31c3d67894d48

With the above change, now compared to 8.3.471:

* `MAIN_MAKE_WINDOW` is unset: https://github.com/RTimothyEdwards/magic/blob/fe9ca3a4ce6544e803db900e0744e9eefd3b7c46/utils/main.c#L349-L350 
* then `MakeMainWindow` is unset and then `MainFileName` is unset : https://github.com/RTimothyEdwards/magic/blob/fe9ca3a4ce6544e803db900e0744e9eefd3b7c46/utils/main.h#L70  https://github.com/RTimothyEdwards/magic/blob/fe9ca3a4ce6544e803db900e0744e9eefd3b7c46/utils/main.c#L375-L378 (With 8.3.471 `MainFileName` is set as `./ccd`, while with 8.3.476 now `MainFileName` is `(null`)
* then with 8.3.476 `TechDefault` is unset (null), while with 8.3.471 `TechDefault` is set as `scmos` expectedly: https://github.com/RTimothyEdwards/magic/blob/fe9ca3a4ce6544e803db900e0744e9eefd3b7c46/utils/main.c#L1124-L1125
* then with 8.3.476 `scmos` technology file is not loaded, while with 8.3.471 `scmos` technology file is loaded expectedly: https://github.com/RTimothyEdwards/magic/blob/fe9ca3a4ce6544e803db900e0744e9eefd3b7c46/utils/main.c#L1131-L1134

Is 8.3.472 and above behavior expected?

RTimothyEdwards, 2024-05-03:
No, this is unintended behavior.  It appears that the "-nowindow" argument, inserted by the GUI wrapper code, was previously being ignored, but I need to figure out what the intended implementation is supposed to be.  Thank you for bringing it to my attention.

RTimothyEdwards, 2024-05-04:
The breaking of backwards-compatibility was unintentional, although there are good reasons for breaking it.  Your use case involves not passing a tech file to magic with the "-T" argument, so magic has to infer the technology file from the file contents.  But historically, all old magic layouts used the same technology name "scmos" which made it impossible to distinguish between incompatible techologies.  I was moving the code to a smarter method using the open_pdks tech file conventions.  But retaining backwards compatibility for the older technology files wasn't difficult to implement, so I went ahead and did so.

Thank you for the detailed report, which was very helpful for debugging the issue.

The corrected code is in version 8.3.477 (update will mirror to github in about 5 hours).

mtasaka, 2024-05-04:
Okay, I've confirmed that 8.3.477 is working as expected, thank you!

closed 2024-05-04 by mtasaka