RTimothyEdwards/magic #307: Regression in DRC detection

donn, 2024-05-12:
Magic version `8.3.466` correctly finds DRC issues in a gds file, but `8.3.479` reports no DRC errors. Are we missing something…?

[reproducible.tgz](https://github.com/RTimothyEdwards/magic/files/15285564/reproducible.tgz)

d-m-bailey, 2024-05-12:
@RTimothyEdwards `8.3.471` is the commit that stops extracting errors. 
`8.3.470` shows
```
Magic 8.3 revision 470 - Compiled on Sun May 12 06:24:50 PDT 2024.
...
DRC style is now "drc(full)"
Loading DRC CIF style.
[INFO] COUNT: 171
```
`8.3.471`
```
Magic 8.3 revision 471 - Compiled on Sun May 12 06:27:30 PDT 2024.
...
DRC style is now "drc(full)"
Loading DRC CIF style.
No errors found.
[INFO] COUNT: 0
```

RTimothyEdwards, 2024-05-12:
The only thing I changed in 8.3.471 is the parsing of arguments to the "select" command.  My best guess is that in fixing the parsing of some of the "select" syntax that was broken, I broke something else.  Is there a critical "select" command in the DRC script?

RTimothyEdwards, 2024-05-12:
If I run the script commands individually, then the issue is clearly with the "select" command since "select top cell" returns "PHY_EDGE_ROW_1_Left_14" and not "inverter".  Not sure why yet---I ran other quick tests and "select top cell" appeared to be doing the right thing.

RTimothyEdwards, 2024-05-12:
The problem seems to be specific to using "magic -dnull".

RTimothyEdwards, 2024-05-12:
Version 8.3.482 should fix this.  The "select top cell" was behaving like "select cell" because there was no specific code handling the "top" option.  The change to be more systematic about handling the various arguments like "more", "less", etc., caused this issue to suddenly manifest itself.  The new code simply registers the fact that "top" has been issued and selects the top level cell as it should.

I have tested this on the reproducible case and it works.

donn, 2024-05-19:
Confirmed fixed. Thanks, Tim!

closed 2024-05-19 by donn