RTimothyEdwards/magic #319: New feedback warning related to diodes in regression test

d-m-bailey, 2024-08-05:
Magic 8.3.478 

Possibly related to open_pdk `1.0.471` commit `97d0844` on 2024.03.09 

The new warning is 
```
feedback add "Could not determine device boundary" pale
```

The relevant code seems to be in `extract/ExtBasic.c`
```
2426                 /*------------------------------------------------------*/
2427                 /* Note that the tr_termvector says a lot about the     */
2428                 /* device geometry.  If the sum of x and y for any      */
2429                 /* vector is 0, then the terminal is enclosed (annular  */
2430                 /* device).  If the sum of x and y for all vectors is   */
2431                 /* zero, then we have a normal rectangular device.  But */
2432                 /* if the sum of all x and y is nonzero, then the       */
2433                 /* device length changes along the device (including    */
2434                 /* bends).  This is a trigger to do a more extensive    */
2435                 /* boundary search to find the exact dimensions of the  */
2436                 /* device.                                              */
2437                 /*------------------------------------------------------*/
2438 
2439                 if (n == 0)
2440                 {
2441                     /* Don't issue a warning on devices such as a       */
2442                     /* vertical diode that may declare zero terminals   */
2443                     /* because the default substrate node is the other  */
2444                     /* terminal.                                        */
2445 
2446                     if (ExtDoWarn && (devptr->exts_deviceSDCount > 0))
2447                         extTransBad(def, reg->treg_tile,
2448                                 "Could not determine device boundary");
2449                     length = width = 0;
2450                 }
```

LVS is passing and the diodes appear to be extracted, so I'm treating this as non-critical (maybe no issue).

RTimothyEdwards, 2024-08-07:
@d-m-bailey :  Is there an indication of _which_ diode device resulted in this error?  Because the code (see the comment) was supposed to avoid generating an error message for diodes.  I might have just missed a use-case, but that seems unlikely since all the diodes are extracted in the same way.  The only device that might violate that is the Schottky diode from the GF180MCU PDK.

d-m-bailey, 2024-08-07:
@RTimothyEdwards `sky130_fd_pr__diode_pw2nd_05v5` and `sky130_fd_pr__diode_pw2nd_11v0` generate warnings. It appears that `sky130_fd_pr__diode_pd2nw_11v0` does not.