RTimothyEdwards/magic #321: obsm* conflict with metal* on gds read (sky130)

d-m-bailey, 2024-08-11:
No error with magic `8.3.465` and open_pdks `1.0.470`
With magic `8.3.478` and open_pdks `1.0.488`
there are many errors like these in the gds read feedback
```
box 600480 580159 600800 580240
feedback add "Illegal overlap between obsm4 and metal4 (types do not connect)" medium
box 599792 580159 599800 580222
feedback add "Illegal overlap between obsm3 and metal3 (types do not connect)" medium
box 599728 580159 599792 580222
feedback add "Illegal overlap between obsm3 and via3 (types do not connect)" medium
```

This is because, for example, `obsm3` and `metal3` are defined on the same plane, and 2 shapes can not overlap on the same plane, correct?

`obsm*` and `metal*` often coexist (at least cross-hierarchy - macros are sometimes covered with `obsm*`), so maybe the `obsm*` should not be defined as existing on a plane.

d-m-bailey, 2024-08-11:
Having trouble recreating the non-issue version that extracted the following
```
.subckt QF_U5_skullfet_inverter A Y VPWR VGND
X0 VGND A Y sky130_fd_pr__nfet_01v8 ad=4.2687 pd=10.82 as=6.4314 ps=26.72 w=4.05 l=0.4
X1 Y A VPWR sky130_fd_pr__pfet_01v8 ad=6.2694 pd=26.64 as=4.4307 ps=10.9 w=4.05 l=0.4
.ends
```

the current extraction is 
```
.subckt QF_U5_skullfet_inverter     a_180_2640# a_250_570#
X0 a_180_2640# a_20_490# a_250_570# sky130_fd_pr__nfet_01v8 ad=4.2687 pd=10.82 as=6.4314 ps=26.72 w=4.05 l=0.4
X1 a_250_570# a_20_490# w_130_100# sky130_fd_pr__pfet_01v8 ad=6.2694 pd=26.64 as=4.4307 ps=10.9 w=4.05 l=0.4
.ends
```

d-m-bailey, 2024-08-12:
This is appears to be due to an open_pdks change.

However, the root cause may be that `obsm*` is stored on an actual data plane and prevents labels from connecting to underlying actual shapes.

Will add test case.

d-m-bailey, 2024-08-12:
[test_obs.tgz](https://github.com/user-attachments/files/16576595/test_obs.tgz)
```
tar xzf test_obs.tgz
cd test_obs
./run_test
```
You should be able to see the difference between open_pdks `1.0.470` and `1.0.488`. Using `obsm*` now breaks extraction.
```
< .subckt QF_U5_skullfet_inverter A Y VPWR VGND
< X0 VGND A Y VGND sky130_fd_pr__nfet_01v8 ad=4.2687 pd=10.82 as=6.4314 ps=26.72 w=4.05 l=0.4
< X1 Y A VPWR VPWR sky130_fd_pr__pfet_01v8 ad=6.2694 pd=26.64 as=4.4307 ps=10.9 w=4.05 l=0.4
---
> .subckt QF_U5_skullfet_inverter A Y VPWR VGND a_180_2640# a_250_570#
> X0 a_180_2640# a_20_490# a_250_570# a_180_2640# sky130_fd_pr__nfet_01v8 ad=4.2687 pd=10.82 as=6.4314 ps=26.72 w=4.05 l=0.4
> X1 a_250_570# a_20_490# w_130_100# w_130_100# sky130_fd_pr__pfet_01v8 ad=6.2694 pd=26.64 as=4.4307 ps=10.9 w=4.05 l=0.4
```

RTimothyEdwards, 2024-08-12:
@d-m-bailey :  I decided that I had to support fill block on all metal layers separately so I assigned the `obsm*` layers to serve that purpose, since otherwise `obsm*` layers are used only in LEF for obstructions and are not found in GDS.

One question would be why is there fill block in this cell?

I do not think that fill block in a separate cell would interfere with pin extraction, though.

RTimothyEdwards, 2024-08-12:
@d-m-bailey :  The test case has obsm1...obsm5 in the same cell as the metal and the fill obstruction layers do not overlap any metal layers.  The ports of the cell are outside of the obstruction layers, which are (appropriately) limited to an area that is meant to be seen from above without being obstructed by fill layers.  The obstruction layers should have no impact on the extraction.

RTimothyEdwards, 2024-08-12:
@d-m-bailey :  Oh, I looked at the layout of "skullfet_nand" but that does not have an issue in extraction, only the "skullfet_inverter".  And I can see that there were shapes on metal1 that got obliterated by the fill on the same plane.
And it is clearly caused by my commenting out the part in the tech file that says "Do not infer generic 'fillblock' from GDS input".
It can probably be solved by doing an "and-not" with the metal layer on the same plane when reading in the fill block types from GDS;  I'm not sure if that's the actual solution I want, but it would fix this error.

RTimothyEdwards, 2024-08-12:
I think putting the fill block in a separate cell would have prevented the extraction error and would only produce an error message with no actual consequences.

RTimothyEdwards, 2024-08-12:
Confirmed that my proposed solution solves the problem, so I will implement that for now.

RTimothyEdwards, 2024-08-12:
Fixed in open_pdks, as this is not an issue with magic, per se.

RTimothyEdwards, 2024-12-26:
This issue is presumed to be resolved, so I'm closing the ticket.

closed 2024-12-26 by RTimothyEdwards

d-m-bailey, 2024-12-26:
@RTimothyEdwards thanks. I'll use the new version of magic and check the regression tests for the next LVS update.